MarlinFirmware/Marlin #14934: Co2 Raster Laser On Marlin 2.0 on Re-Arm

mattr254, 2019-08-12:
I've been running Marlin on my Mega 2560 for over a Year now using Hardware PWM's To control the Intensity and enable and it works great on raster images. Well, it's slow but it still works (Max Speed 3000 mm/s). I'm able to get this to work by enabling fastPWM and using fan commands.

I've been trying to use the new 32 bit F/W on the Re-arm and can't get it to work. The Pins have output. I've tried several of the PWM pins on the board running a break out not a shield. The laser doesn't seem to be in sync with the movement. I've tried both the interrupt type and the dedicated type. 
I've tried Pins P1_18,P2_04,P1_21 All in different combinations for both intensity and enable.

Am I missing something?

p3p, 2019-08-12:
Can you confirm you used the same version of Marlin on both boards, we need to verify that the change in functionality is caused by the mcu, not Marlin version.

mattr254, 2019-08-12:
I'm running 1.1.9 on the mega 2560, I'll load up 2.0 tonight and reply back
with my findings. Thank you for the quick reply back!

mattr254, 2019-08-13:
So per .p3p's request, I loaded 2.0 into the mega 2560. I got the same results that I got with installing it into the re-arm. The laser output is not synced to the movement in raster mode. The pins are the exact same that I had when I ran 1.1.9 successfully. I didn't think to do that first, It was a great idea! 

A few months back I had the same issue with a teensy 3.2 with Marlin 2.0 and it had the same exact issue. I wrote it off as it not supporting fast PWM so I went back to my trusty Arduino Mega running 1.1.9 and sidelined the teensy. I had a controller die and figured why not upgrade to a re-arm?

p3p, 2019-08-13:
Had a quick look through the open issues regarding lasers and found #11576, you are probably hitting the same issue as them, a limit was added on how often `planner.check_axes_activity();` can be called.

You could trying doing undoing this change although the commit it was added in didn't explain why the limit was added.

Change Marlin.cpp @ line 636
```
  // Limit check_axes_activity frequency to 10Hz
  static millis_t next_check_axes_ms = 0;
  if (ELAPSED(ms, next_check_axes_ms)) {
    planner.check_axes_activity();
    next_check_axes_ms = ms + 100UL;
  }
``` 
to just
```
planner.check_axes_activity();
```

p3p, 2019-08-13:
Found the PR(#8446) it was changed in and ironically it was added because of a limitation in the LPC176x Software PWM system, the implementation has changed since then (late 2017) to not be blocking so I'm not sure it is still needed at all ..

mattr254, 2019-08-13:
.p3p Thank you for looking into this, I found the lines of code and will make the changes and update this post. Also looking into #11576, very good lead!!!

mattr254, 2019-08-14:
I updated the code to reflect and it works! I'm going to post this in the laser spindle section for others Thank you again!

p3p, 2019-08-14:
I'm pretty sure the update frequency limit is not required anymore, for the reason it was added at least, does it run ok on the LPC176x(Re-ARM)?

mattr254, 2019-08-14:
Yes it does, I tested it last night. :)

boelle, 2019-09-24:
@mattr254 so we can close this one?

Thunder8888, 2020-02-26:
> I've been running Marlin on my Mega 2560 for over a Year now using Hardware PWM's To control the Intensity and enable and it works great on raster images. Well, it's slow but it still works (Max [...]

Hey - i have the same setup with Marin 1.1.9 and the Atmega2560. I enabled Fan Control and Fast_PWM. But the engraving still is not in sync. May you help me ?!

TY!

mattr254, 2020-02-26:
Your intensity and trigger wires might be backward. I found the best pins
to use are the servo ctrl pins. I moved over to Somoothieware on my Re-arm
and haven't looked back so my memory is a little foggy on the Marlin
specifics... On Smoothieware I like how it is able to automatically stop
the laser output when the head stops and switches directions so it doesn't
burn hotter in the corners. Sommthieware doesn't use fan control instead it
uses G1 and G0 to turn the laser on and off. If you're going to do a lot of
raster images I would recommend it. Vector burns Atmega should be able to
handle it.

Thunder8888, 2020-02-26:
Ok - i‘ll try! Ty! Which board / software do you use atm /w smoothieware

mattr254, 2020-02-26:
Re-ARM it's a direct RAMPS 1.4 Plugin

thinkyhead, 2020-02-28:
Revisions to laser code are in the works, but still awaiting detailed test feedback.
#15335 and #16082. It would be good to get both sets of changes in, but we need them to fit together better.