MarlinFirmware/Marlin #14937: UNKNOWN_Z_NO_RAISE not working with positive z offset

kazolar, 2019-08-13:
### Description

enable UNKNOWN_Z_NO_RAISE 

1. enable UNKNOWN_Z_NO_RAISE 
2. set Z offset to positive (needed for piezo probing)
3. Enable Z_SAFE_HOMING 
4. Try to home 

**Expected behavior:** 
Expect it home X,Y then move to center and move the bed up toward the nozzle with no reverse motion

**Actual behavior:** 
XY home, move to center then the bed tries to move down (in positive direction) and it's hitting the bottom triggering the piezo as a result and doesn't home Z

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-08-13:
Include a ZIP file containing your Configuration.h and Configuration_adv.h files.

kazolar, 2019-08-13:
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3498242/config_files.zip)
Attached.
I enabled debug, and I am not seeing "Raise Z (before homing) to" issued, but the machine still tries to send the bed through the floor. I am guessing it has something to do with movement to make space for the probe -- but that should not happen for z homing when then bed fell all the way down.

Roxy-3D, 2019-08-13:
Turn on 
```
//#define DEBUG_LEVELING_FEATURE
```
Then issue a M111 S255     And home the machine.   Paste that verbose log here.   That will give us the information needed to figure out what is wrong.

kazolar, 2019-08-13:
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
ok
echo:G28
>>> G28
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X:0 Y:0 Z:0.00 (Aligned With & Same Z as Nozzle)
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -307.50, [50.00])
echo:busy: processing
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, [50.00])
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position=(0.00, 0.00, 0.00) : 
<<< set_axis_is_at_home(X)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -382.50, [50.00])
echo:busy: processing
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, [50.00])
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position=(0.00, 0.00, 0.00) : 
<<< set_axis_is_at_home(Y)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
Z_SAFE_HOMING >>>
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  destination=(102.00, 127.00, 0.00) : Z_SAFE_HOMING
>>> do_blocking_move_to(102.00, 127.00, 0.00)
<<< do_blocking_move_to
echo:busy: processing
>>> homeaxis(Z)
  current_position=(102.00, 127.00, 0.00) : set_probe_deployed
deploy: 1
>>> do_blocking_move_to(102.00, 127.00, 0.00)
<<< do_blocking_move_to
Home 1 Fast:
>>> do_homing_move(Z, -300.00, [5.00])
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 5.00, 5.00)
echo:busy: processing
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 5.00)
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
Axis Z home_offset = 0.00 position_shift = 0.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = 0.00
> home_offset[Z] = 0.00
  current_position=(102.00, 127.00, 0.00) : 
<<< set_axis_is_at_home(Z)
  current_position=(102.00, 127.00, 0.00) : sync_plan_position
  current_position=(102.00, 127.00, 0.00) : > AFTER set_axis_is_at_home
  current_position=(102.00, 127.00, 0.00) : set_probe_deployed
deploy: 0
>>> do_blocking_move_to(102.00, 127.00, 0.00)
<<< do_blocking_move_to
<<< homeaxis(Z)
<<< Z_SAFE_HOMING
  current_position=(102.00, 127.00, 0.00) : sync_plan_position
X:102.00 Y:127.00 Z:0.00 E:0.00 Count X:9066 Y:11288 Z:0
<<< G28
ok

nothing in here that says that it's trying to move down prior to homing Z, but it is -- I even set the probe offset to 0, and it still tries to bang the bed into the floor

Roxy-3D, 2019-08-13:
Can you highlight the line where things go sour?   Most of the log is OK, Right?   The X & Y homing part of the log are correct, Right?

(But I do see what you mean!   The log doesn't show that it knows it is going sub-zero!)

kazolar, 2019-08-13:
I added 
#define DELAY_BEFORE_PROBING 300
200 -- which is default wasn't enough for the piezo to survive the rattling of it trying to hit the bed into the floor, so it false triggered right away
the problem occurs right after it does the safe Z homing, it moves to center, then tries to move down. The log doesn't show it, so I'm pretty confused why it does it. I was expecting to see
"Raise Z (before homing) to" -- the spot it does that in G28.cpp, but that's not the culprit -- it's something else. 

and now it survives the banging into the ground -- and continues with the probing
here is what happens now. It's still not good that it's trying to bang the bed into the floor, as it doesn't make sense to do it, and it's rather unnerving.  

```
echo:DEBUG:ECHO,INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING
ok
echo:G28
>>> G28
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X:0 Y:0 Z:0.10 (Aligned With & Above Nozzle)
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -307.50, [50.00])
echo:busy: processing
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, [50.00])
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
Axis X home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position=(0.00, 0.00, 0.00) : 
<<< set_axis_is_at_home(X)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -382.50, [50.00])
echo:busy: processing
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, [50.00])
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
Axis Y home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position=(0.00, 0.00, 0.00) : 
<<< set_axis_is_at_home(Y)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
Z_SAFE_HOMING >>>
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  destination=(102.00, 127.00, 0.00) : Z_SAFE_HOMING
>>> do_blocking_move_to(102.00, 127.00, 0.00)
<<< do_blocking_move_to
echo:busy: processing
>>> homeaxis(Z)
  current_position=(102.00, 127.00, 0.00) : set_probe_deployed
deploy: 1
>>> do_blocking_move_to(102.00, 127.00, 0.00)
<<< do_blocking_move_to
Home 1 Fast:
>>> do_homing_move(Z, -300.00, [5.00])
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 5.00, 5.00)
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -10.00, 5.00)
echo:busy: processing
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
Axis Z home_offset = 0.00 position_shift = 0.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = 0.10
> home_offset[Z] = 0.00
  current_position=(102.00, 127.00, -0.10) : 
<<< set_axis_is_at_home(Z)
  current_position=(102.00, 127.00, -0.10) : sync_plan_position
  current_position=(102.00, 127.00, -0.10) : > AFTER set_axis_is_at_home
  current_position=(102.00, 127.00, -0.10) : set_probe_deployed
deploy: 0
>>> do_blocking_move_to(102.00, 127.00, -0.10)
<<< do_blocking_move_to
<<< homeaxis(Z)
<<< Z_SAFE_HOMING
  current_position=(102.00, 127.00, -0.10) : sync_plan_position
X:102.00 Y:127.00 Z:-0.10 E:0.00 Count X:9066 Y:11288 Z:-40
<<< G28
ok
```

Roxy-3D, 2019-08-13:
> The log doesn't show it, so I'm pretty confused why it does it. I was expecting to see
"Raise Z (before homing) to" -- the spot it does that in G28.cpp, but that's not the culprit -- it's something else.

It is possible something got moved or deleted...    I'll take a look at this code as soon as I get a few minutes.    (And very possibly...  add or replace those debug lines.)

kazolar, 2019-08-14:
So this is strange, after setting the DELAY_BEFORE_PROBING 300 it no longer is trying to send the bed through the floor and working correctly -- very strange...I'll keep testing

Roxy-3D, 2019-08-14:
This problem may not be as simple as you think.   Some of the bed leveling systems (UBL for example) used to have problems when you went 'off the mesh' or into places where the mesh was not well defined.   

You included your configuration.h files (as you should)....   But save me the effort.   What bed leveling system are you using?    I'm wondering if the problem you are seeing is at a higher level and in the Z-Compensation logic  (Bed leveling).

kazolar, 2019-08-14:
When I was seeing the issue, I hadn't setup bed compensation yet. I have last night and I am using linear -- flat bed, adjust for tilt is all that's needed. I am going to guess as to what was triggering weird z behavior, I haven't studies the code if it does this, but if when it starts to home z, and z probe is already triggered, would it reverse some distance to "un-trigger" it -- that's what looked like it was doing -- home x, y -- move to center -- and without a sufficient pause there, the piezo is triggered and it then tries to lower the bed -- send Z into positive direction (instead of failing as I believe it should since z is already triggered) -- with a pause -- I had to increase the pause to 700 ms for the purposes of G29 as it was false triggering with a lower pause, so now it pauses, piezo is not triggered and it just proceeds to raise the bed. Definitely the movement to release the probe which was hammering the bed into the floor was not logged in debug though, but with the pause it's no longer trying to do that and working perfectly -- with both G28 and G29, This is MakerGear M2 I rebuilt the extruder with my own machined mounting plates for a bondtech bmg and am using an e3d v6 with a nozzle x and Precision Piezo Orion. Love the M2, don't like the limitation of the extruder mount, also went ahead and swapped the board for SKR1.3 -- with tmc2209s -- now working correctly all that's left is PID tune the hotend. Piezo (when right) makes leveling really simple offset is tiny and it's going to be such that it will trigger just above the point of the hotend -- so I have it set to 0.1 here, I've had it anywhere from 0.05 to 0.1 on different machines. Thanks for the help. If you see the movement I'm talking about, adding it to leveling debug would help.

closed 2019-08-14 by kazolar