MarlinFirmware/Marlin #14938: TMCStepper update to 0.4.6

shaggyjoseph, 2019-08-13:
### Description
TMCStepper update to 0.4.6,compile error.

### Steps to Reproduce

1.what TMCStepper.h change??
2. #define SW_CAPABLE_PLATFORM defined(__AVR__) || defined(TARGET_LPC1768) || defined(ARDUINO_ARCH_ESP32) || defined(ARDUINO_ARCH_STM32F1)
so it need #include <SoftwareSerial.h> file
3. delete  defined(ARDUINO_ARCH_STM32F1)

**Expected behavior:** correctly error

**Actual behavior:** no SoftwareSerial.h file

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tpruvot, 2019-08-13:
see #14892

shaggyjoseph, 2019-08-13:
when i delete defined(ARDUINO_ARCH_STM32F1),it can correctly compile!

boelle, 2019-08-13:
Include a ZIP file containing your Configuration.h and Configuration_adv.h files.

shaggyjoseph, 2019-08-15:
[my_config.zip](https://github.com/MarlinFirmware/Marlin/files/3504484/my_config.zip)

tpruvot, 2019-08-15:
This problem was already fixed this week in PR #14861 (Aug 8)

shaggyjoseph, 2019-08-15:
Thanks! @tpruvot

closed 2019-08-15 by shaggyjoseph