MarlinFirmware/Marlin #14939: [env:STM32F1] lib_ignore   = U8glib-HAL can cause compilation errors

shaggyjoseph, 2019-08-13:
### Description
[env:STM32F1] lib_ignore   = U8glib-HAL ，can ignore u8g lib,so cause compilation errors.the errors is "conflicting types for 'random'";can not distinguish function overloading!

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-08-13:
Include a ZIP file containing your Configuration.h and Configuration_adv.h files.

shaggyjoseph, 2019-08-15:
[my_config.zip](https://github.com/MarlinFirmware/Marlin/files/3504270/my_config.zip)

tpruvot, 2019-08-15:
What is exactly your board/chip ?

shaggyjoseph, 2019-08-16:
stm32f103RET6

boelle, 2019-09-24:
@niu3niu still having problems?

niu3niu, 2019-09-25:
@boelle  i have solve it.Thanks ! u can close this issue!

closed 2019-09-25 by boelle