MarlinFirmware/Marlin #14942: Several problems with Rumba32 (Stm32F446)

MarkMan0, 2019-08-13:
I want to upgrade my Ender 3 with a Rumba32 board, which uses a powerful STM32 processor.

The first problem is, that I cannot compile in platformIO. 
For some reason it can't find the TMCStepper.h header, but the library is installed to global storage.  Compiling for the SKR V1.3, everything is working, same directory, different branches, same config.
There is another error `SerialUsb was not declared`.

So based on [this](https://github.com/MarlinFirmware/Marlin/issues/14295#issuecomment-502412351) issue, I switched to the Arduino extension in vscode, and installed the TMCStepper library, and the dev branch of this [MarlinFirmware/U8glib-HAL](https://github.com/MarlinFirmware/U8glib-HAL/tree/dev) library.

With this configuration I can sucessfully compile the latest Marlin 2.0.

The problem is, that there are problems with the LCD - Reprap discount Full Graphic Controller and the firmware.

The first one being - the firmware won't boot, only when I connect to the board through serial will it boot. After connecting the power(24 V) the onboard LED doesn't start flashing. Only after I attach a serial monitor will it boot up, display the Marlin logo on the LCD and start blinking the LED. 
The onboard 5V regulator is working properly, because after disconnecting the USB, I can still interract with the board.

2nd issue:
After it boots up, the characters on the LCD are scattered. I can see it's trying to display what it's supposed to, but it looks like there is some kind of communication issue - random pixels lighting up, or turning off. I tested the LCD with the SKR board, and it was working properly, so the issue is on the rumba board. Also tried different cables - didn't solve the problem.

Below is attached my Marlin config and the Arudino libraries folder( `_username_\Documents\Arduino\libraries`
[libraries.zip](https://github.com/MarlinFirmware/Marlin/files/3496576/libraries.zip)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3496587/Marlin.zip)

For compiling, you will need to follow [this](https://github.com/Aus3D/RUMBA32/wiki/Getting-RUMBA32-Up-&-Running) guide. You also need to unzip the libraries.zip file to you libraries folder.

MarkMan0, 2019-09-27:
I'm away from my printer, can test in a week or 2 only

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle