MarlinFirmware/Marlin #14943: [BUG] Cannot compile when setting TMC2208 as driver for MKS Gen L board

Alex-Angee, 2019-08-13:
### Description

When setting TMC2208 as the stepper driver, I get multiple errors compiling.

### Steps to Reproduce

1. Include libraries TMCStepper, TMC2208Stepper, and TMC2130Stepper
2. Define MKS as the printer
3. Set TMC2208 as the stepper driver

**Expected behavior:** Compiled successfully

**Actual behavior:** R_SENSE not declared in this scope, HOLD_MULTIPLIER not declared in this scope.

####Additional information: This is with the firmware provided by TH3DStudio using Marlin.

####Error Log
```
sketch\stepper_indirection.cpp: In function 'void tmc2208_init(TMC2208Stepper&, uint16_t, uint16_t, uint32_t, float)':

stepper_indirection.cpp:460:24: error: 'HOLD_MULTIPLIER' was not declared in this scope

     st.rms_current(mA, HOLD_MULTIPLIER, R_SENSE);

                        ^

stepper_indirection.cpp:460:41: error: 'R_SENSE' was not declared in this scope

     st.rms_current(mA, HOLD_MULTIPLIER, R_SENSE);

                                         ^

stepper_indirection.cpp:464:15: error: 'INTERPOLATE' was not declared in this scope

     st.intpol(INTERPOLATE);

               ^

sketch\stepper_indirection.cpp: In function 'void tmc2208_init_to_defaults()':

stepper_indirection.cpp:494:21: error: 'X_CURRENT' was not declared in this scope

       _TMC2208_INIT(X, planner.axis_steps_per_mm[X_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:61: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                             ^

stepper_indirection.cpp:494:21: error: 'X_MICROSTEPS' was not declared in this scope

       _TMC2208_INIT(X, planner.axis_steps_per_mm[X_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:75: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                           ^

stepper_indirection.cpp:494:21: error: 'X_HYBRID_THRESHOLD' was not declared in this scope

       _TMC2208_INIT(X, planner.axis_steps_per_mm[X_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:92: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                                            ^

stepper_indirection.cpp:500:21: error: 'Y_CURRENT' was not declared in this scope

       _TMC2208_INIT(Y, planner.axis_steps_per_mm[Y_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:61: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                             ^

stepper_indirection.cpp:500:21: error: 'Y_MICROSTEPS' was not declared in this scope

       _TMC2208_INIT(Y, planner.axis_steps_per_mm[Y_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:75: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                           ^

stepper_indirection.cpp:500:21: error: 'Y_HYBRID_THRESHOLD' was not declared in this scope

       _TMC2208_INIT(Y, planner.axis_steps_per_mm[Y_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:92: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                                            ^

stepper_indirection.cpp:506:21: error: 'Z_CURRENT' was not declared in this scope

       _TMC2208_INIT(Z, planner.axis_steps_per_mm[Z_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:61: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                             ^

stepper_indirection.cpp:506:21: error: 'Z_MICROSTEPS' was not declared in this scope

       _TMC2208_INIT(Z, planner.axis_steps_per_mm[Z_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:75: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                           ^

stepper_indirection.cpp:506:21: error: 'Z_HYBRID_THRESHOLD' was not declared in this scope

       _TMC2208_INIT(Z, planner.axis_steps_per_mm[Z_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:92: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                                            ^

stepper_indirection.cpp:512:21: error: 'E0_CURRENT' was not declared in this scope

       _TMC2208_INIT(E0, planner.axis_steps_per_mm[E_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:61: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                             ^

stepper_indirection.cpp:512:21: error: 'E0_MICROSTEPS' was not declared in this scope

       _TMC2208_INIT(E0, planner.axis_steps_per_mm[E_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:75: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                           ^

stepper_indirection.cpp:512:21: error: 'E0_HYBRID_THRESHOLD' was not declared in this scope

       _TMC2208_INIT(E0, planner.axis_steps_per_mm[E_AXIS]);

                     ^

sketch\stepper_indirection.cpp:490:92: note: in definition of macro '_TMC2208_INIT'

   #define _TMC2208_INIT(ST, SPMM) tmc2208_init(stepper##ST, ST##_CURRENT, ST##_MICROSTEPS, ST##_HYBRID_THRESHOLD, SPMM)

                                                                                            ^

exit status 1
'HOLD_MULTIPLIER' was not declared in this scope

[Config files with EZABL.zip](https://github.com/MarlinFirmware/Marlin/files/3497326/Config.files.with.EZABL.zip)

teemuatlut, 2019-08-14:
> This is with the firmware provided by TH3DStudio using Marlin.

And as such I ask that you bring this up there.

boelle, 2019-09-24:
will close as this is not marlin firmware related