MarlinFirmware/Marlin #14945: X_SERIAL_(RX|TX)_PIN information not recognized - compiler error

kulfuerst, 2019-08-13:
### Description

Marlin doesn't compile. error reads:

```
In file included from c:\users\andr~1\appdata\local\temp\arduino_build_355909\sketch\src\inc\marlinconfig.h:41:0,

from sketch\src\HAL\HAL_AVR\HAL.cpp:29:

c:\users\andr~1\appdata\local\temp\arduino_build_355909\sketch\src\inc\SanityCheck.h:1822:4: error: #error "TMC2208 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."

#error "TMC2208 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
```

### Steps to Reproduce

1. Check out my files [here on Github](https://github.com/kulfuerst/Marlin_Ai3M_2.0.x_bugfix_piezo)
2. Install TMCstepper libraries
3. attempt compilation

**Expected behavior:** [What you expect to happen]

The firmware should compile and make me happy

**Actual behavior:** [What actually happens]

The firmware makes me sad. I don't know why the error message appears in spite of all pins being assigned in pins_TRIGORILLA_14.h

Thanks to everyone for your help here. I don't know what to do anymore.

paulpenney, 2019-08-13:
Getting the same issues. Trying to compile for 2209 on a FYSETC F6 board - I can get it to compile for an ATmega2560 on Ramps, but changing to the FYSETC F6 board profile errors out. The pins are assigned properly in the F6 pins file.

paulpenney, 2019-08-13:
Looks like its the way its defined in the pins file. If I replace this in pins_FYSETC_F6_13.h: 

`#if HAS_TMC220x
  /**
   * TMC2208/TMC2209 stepper drivers
   *
   * Software serial communication pins.
   * At the moment, F6 rx pins are not pc interrupt pins
   */
  #define X_SERIAL_RX_PIN  -1   // 71
  #define X_SERIAL_TX_PIN  72
  #define Y_SERIAL_RX_PIN  -1   // 73
  #define Y_SERIAL_TX_PIN  75
  #define Z_SERIAL_RX_PIN  -1   // 78
  #define Z_SERIAL_TX_PIN  79
  #define E0_SERIAL_RX_PIN -1   // 76
  #define E0_SERIAL_TX_PIN 77
  #define E1_SERIAL_RX_PIN -1   // 80
  #define E1_SERIAL_TX_PIN 81
  #define E2_SERIAL_RX_PIN -1   // 22
  #define E2_SERIAL_TX_PIN 82
#endif`

With this from the pins_RAMPS.h

`#if HAS_TMC220x
  /**
   * TMC2208/TMC2209 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial1
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

  //
  // Software serial
  //

  #ifndef X_SERIAL_TX_PIN
    #define X_SERIAL_TX_PIN  40
  #endif
  #ifndef X_SERIAL_RX_PIN
    #define X_SERIAL_RX_PIN  63
  #endif
  #ifndef X2_SERIAL_TX_PIN
    #define X2_SERIAL_TX_PIN -1
  #endif
  #ifndef X2_SERIAL_RX_PIN
    #define X2_SERIAL_RX_PIN -1
  #endif

  #ifndef Y_SERIAL_TX_PIN
    #define Y_SERIAL_TX_PIN  59
  #endif
  #ifndef Y_SERIAL_RX_PIN
    #define Y_SERIAL_RX_PIN  64
  #endif
  #ifndef Y2_SERIAL_TX_PIN
    #define Y2_SERIAL_TX_PIN -1
  #endif
  #ifndef Y2_SERIAL_RX_PIN
    #define Y2_SERIAL_RX_PIN -1
  #endif

  #ifndef Z_SERIAL_TX_PIN
    #define Z_SERIAL_TX_PIN  42
  #endif
  #ifndef Z_SERIAL_RX_PIN
    #define Z_SERIAL_RX_PIN  65
  #endif
  #ifndef Z2_SERIAL_TX_PIN
    #define Z2_SERIAL_TX_PIN -1
  #endif
  #ifndef Z2_SERIAL_RX_PIN
    #define Z2_SERIAL_RX_PIN -1
  #endif

  #ifndef E0_SERIAL_TX_PIN
    #define E0_SERIAL_TX_PIN 44
  #endif
  #ifndef E0_SERIAL_RX_PIN
    #define E0_SERIAL_RX_PIN 66
  #endif
  #ifndef E1_SERIAL_TX_PIN
    #define E1_SERIAL_TX_PIN -1
  #endif
  #ifndef E1_SERIAL_RX_PIN
    #define E1_SERIAL_RX_PIN -1
  #endif
  #ifndef E2_SERIAL_TX_PIN
    #define E2_SERIAL_TX_PIN -1
  #endif
  #ifndef E2_SERIAL_RX_PIN
    #define E2_SERIAL_RX_PIN -1
  #endif
  #ifndef E3_SERIAL_TX_PIN
    #define E3_SERIAL_TX_PIN -1
  #endif
  #ifndef E3_SERIAL_RX_PIN
    #define E3_SERIAL_RX_PIN -1
  #endif
  #ifndef E4_SERIAL_TX_PIN
    #define E4_SERIAL_TX_PIN -1
  #endif
  #ifndef E4_SERIAL_RX_PIN
    #define E4_SERIAL_RX_PIN -1
  #endif
#endif`

It will compile. Looks like it needs the ifndefs.

kulfuerst, 2019-08-13:
Very curious. The code in my pins_RAMPS.h is different in mine it looks like this:

```
#if HAS_DRIVER(TMC2208)
  /**
   * TMC2208 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial1
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

  //
  // Software serial
  //

  #define X_SERIAL_TX_PIN    40
  #define X_SERIAL_RX_PIN    63
  #define X2_SERIAL_TX_PIN   -1
  #define X2_SERIAL_RX_PIN   -1

  #define Y_SERIAL_TX_PIN    59
  #define Y_SERIAL_RX_PIN    64
  #define Y2_SERIAL_TX_PIN   -1
  #define Y2_SERIAL_RX_PIN   -1

  #define Z_SERIAL_TX_PIN    42
  #define Z_SERIAL_RX_PIN    65
  #define Z2_SERIAL_TX_PIN   -1
  #define Z2_SERIAL_RX_PIN   -1

  #define E0_SERIAL_TX_PIN   44
  #define E0_SERIAL_RX_PIN   66
  #define E1_SERIAL_TX_PIN   -1
  #define E1_SERIAL_RX_PIN   -1
  #define E2_SERIAL_TX_PIN   -1
  #define E2_SERIAL_RX_PIN   -1
  #define E3_SERIAL_TX_PIN   -1
  #define E3_SERIAL_RX_PIN   -1
  #define E4_SERIAL_TX_PIN   -1
  #define E4_SERIAL_RX_PIN   -1
#endif
```

I tried using your code to get my pins_TRIGORILLA_14.h to compile, but that also didn't work. In the meantime I've been told in a facebook group that I am not the first with this issue, others have given up.

What I find complicated is that the differences begin right with the very first line. Is it supposed to be "#if HAS_DRIVER(TMC2208)" or "#if HAS_TMC220x"? Whichever one I use, I get the same error though.

paulpenney, 2019-08-13:
I was wrong about my issue. For some reason the RX pins were left undefined. Defining them allowed me to compile.

kulfuerst, 2019-08-13:
So what does your working code look like?

kulfuerst, 2019-08-14:
Issue SOLVED

Marlin does not allow for only TX to be assigned. Assigning empty and unused pins to RX fixed the compiler error.

closed 2019-08-14 by kulfuerst