MarlinFirmware/Marlin #14946: [BUG] References to "SD card" do not make sense with USB_FLASH_DRIVE_SUPPORT

marcio-ao, 2019-08-13:
Currently, the following strings mention a "card" and do not make sense when `USB_FLASH_DRIVE_SUPPORT` is enabled:

```
#define MSG_CARD_MENU                       _UxGT("Print from SD")
#define MSG_NO_CARD                         _UxGT("No SD Card")
#define MSG_SD_UPDATE                       _UxGT("SD Update")
#define MSG_INIT_SDCARD                     _UxGT("Init. SD Card")
#define MSG_CHANGE_SDCARD                   _UxGT("Change SD Card")
#define MSG_RELEASE_SDCARD                  _UxGT("Release SD Card")
#define MSG_SD_INSERTED                     _UxGT("Card Inserted")
#define MSG_SD_REMOVED                      _UxGT("Card Removed")
#define MSG_SD_RELEASED                     _UxGT("Card Released")
```

In our firmware, we typically replace references to "cards" or "SD" with "media" since that is a generic term that covers both cards and USB flash drives.

To keep us from having to do this in the future, I would like to create a PR to make the following changes:

```
#define MSG_MEDIA_MENU                      _UxGT("Print from Media")
#define MSG_NO_CARD                         _UxGT("No Media")
#define MSG_MEDIA_UPDATE                    _UxGT("Media Update")
#define MSG_INIT_MEDIA                      _UxGT("Init. Media")
#define MSG_CHANGE_MEDIA                    _UxGT("Change Media")
#define MSG_RELEASE_MEDIA                   _UxGT("Release Media")
#define MSG_MEDIA_INSERTED                  _UxGT("Media Inserted")
#define MSG_MEDIA_REMOVED                   _UxGT("Media Removed")
#define MSG_MEDIA_RELEASED                  _UxGT("Media Released")
```

For languages other than english, I could change the variable names but leave the existing text the same (translators would have to modify it to use a more generic term later, where applicable).

Anyhow, that's one way to do it. The other way is to define an entirely new set of strings for USB and put `#if ENABLED(USB_FLASH_DRIVE_SUPPORT)`, but I really think that is more complicated than it needs to be.

@thinkyhead: Thoughts?

InsanityAutomation, 2019-08-14:
Ill throw in my vote in support of the simple verbiage change to media. Its what windows shows nowadays as well.

ShriganeshBorse, 2019-08-14:
Looks good suggestion and trivial change

tpruvot, 2019-08-14:
how much boards have usb keys support ?

InsanityAutomation, 2019-08-14:
> 
> 
> how much boards have usb keys support ?

At minimum the archim2 and SKR Pro, and im sure as 32bit is more stable many more will be popping up.

marcio-ao, 2019-08-14:
> At minimum the archim2 and SKR Pro, and im sure as 32bit is more stable many more will be popping up.

Well, technically the Archim 2 does not have external USB flash drive support. The support on our TAZ Pro is provided by the MAX3421E chip via SPI. We've gotten it to work on an 8-bit AVR, but the USB Host Shield library takes up most of the available memory, leaving not very much for Marlin to function properly.

InsanityAutomation, 2019-08-14:
> We've gotten it to work on an 8-bit AVR, but the USB Host Shield library takes up most of the available memory, leaving not very much for Marlin to function properly.

For the next trick, Attiny!!

With that as an SPI addon, it could in theory be added to almost anything so solid reverences to work off of and itll likely start showing up alot more.

AnHardt, 2019-08-14:
The SKR Pro has a plug for a USB-stick - but no software to support it.
--------------
So, up to now, there are no Marlin supported boards being able to handle USB-sticks in the wild?

marcio-ao, 2019-08-14:
@AnHardt: Would [UHS3](https://github.com/felis/UHS30) support it? Although Marlin currently has UHS2, I have gotten UHS3 to work with it and I might push those changes in the future. Anyhow, with UHS3, it should be possible to add new host controllers simply by writing a new host controller interface.

marcio-ao, 2019-08-14:
The SD functionality in Marlin comprises an interface to the SD card physical interface ("Sd2Card.h/cpp") and file system stuff that operates on 512 byte blocks (pretty much everything else in the "sd" directory) The file system stuff works as long as the underlying devices supports 512 byte blocks, which is the case for USB bulk storage class devices. So once you have a USB library that implements USB bulk storage devices, you can get Marlin to read files off it. For UHS, I replaced the Sd2Card.h/cpp with my own classes that forwarded block access to the UHS library instead of to an SD card.

It looks like the SKR Pro is using STM chips. If there is any library or API out there that lets you read and write 512 byte blocks from USB Mass Storage Devices, then it would be relatively straightforward to get Marlin talking to it.

marcio-ao, 2019-08-14:
The following thread may be helpful for anyone wishing to experiment with the current USB functionality:

https://github.com/MarlinFirmware/Marlin/issues/12595

closed 2019-08-14 by thinkyhead