MarlinFirmware/Marlin #14947: Won't Auto home after a print, just chatters[BUG] (short description)

Loadie150, 2019-08-14:
-->

### Description

After completing a print I try to home the printer, All it does is go in the opposite direction and chatters. I can use Pronterface, CURA 4.2.1 and the LCD screen to move the print head around; however, it won't autohome for me.

### Steps to Reproduce

It's periodic at best. I have no idea when it will happen.

**Expected behavior:** [Autohome]

**Actual behavior:** [Z lowers as expected, Y moves to 0 but X moves in the 

opposite direction and chatters while doing it. I have a short video but with my ISP it would take about an hour to upload.]

I've tried to reboot the printer, board, computer, same results. The only solution I've found so far is to re-flash Marlin 2.0 onto the board.

I'm using an SKR v1.3 board with TMC2208's in UART mode. This has happened to me a few times now.

#### Additional Information

I only have winrar right now, I tried uploading that but it wouldn't allow me.

I've uploaded the video to the Marlin facebook page https://www.facebook.com/groups/3Dtechtalk/
https://www.facebook.com/dan.fraser.104/videos/10156061978245683/

boelle, 2019-08-14:
you forgot to attach your 2 config files as zip

boelle, 2019-09-24:
@Loadie150 you forgot to attach your 2 config files as zip

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

larsleukert, 2020-02-12:
@Loadie150 
I have had the same problem for a long time. Today i realised: only with Cura!
I have a G91 in my end script to move relative away from the printed part.
Then: cura does not do a G90 at the begin of an print - but simplyfy3D does.
Puting a G90 at the end of the endscript solved the problem for me.
I think it would also work in the start script