MarlinFirmware/Marlin #14948: [BUG] [bugfix-2.0.x] Disabling SEGMENT_LEVELED_MOVES causes hard crashes

orcinus, 2019-08-14:
### Steps to Reproduce

1. Disable (comment out) SEGMENT_LEVELED_MOVES
2. Enable MBL
3. Home All
4. Enable bed leveling

**Expected behavior:**

No crash.

**Actual behavior:**

Hard crash, AVR completely unresponsive, LED1 flashing, reset button does not work, need to power cycle.

boelle, 2019-08-14:
attach your 2 config files as zip

orcinus, 2019-08-14:
Are you able to reproduce the crash by following the steps?

orcinus, 2019-08-14:
Config files attached, not that they’re required to reproduce the issue.
[BigBox.zip](https://github.com/MarlinFirmware/Marlin/files/3503667/BigBox.zip)

Print init G script is as follows:
```
G90
M83
M106 S0

;BigBox Printer Start Script Begin
;relies on M83 (use relative extrusion distances)

M117 Preparing

;home
T0 ; select extruder 0
G28
M420 S1

;set values
; M218 T1 X-37.9 Y1.6 ; set extruder 1 offset

;move to dock
T0 ; select extruder 0
G1 Z4 F3000 ; lift the head
G1 X100 Y200 F3000 ; move to position in front of dock
;G1 X100 Y240 F3000 ; move into dock

;heat bed
M190 S<BED> ; wait for bed temperature

;heat extruder 0
T0 ; select extruder 0
M109 S<TEMP> ; set extruder 0 temperature and wait

M420 S1

M117 Print starting

;BigBox Printer Start Script End
```

The crash occurs at the first M420 S1 line.

Roxy-3D, 2019-08-14:
You have Mesh Bed Leveling enabled...   Can you try a different bed leveling system?
Either Bi-Linear or UBL would be a good data point.

orcinus, 2019-08-14:
Sure, I'll give it a go as soon as i'm done with the current batch of prints.

Do note that i've had issues with UBL before (it's an IDEX machine) so that might not be a very valid data point. Although i guess it'll serve fine just to narrow down the crash.

Roxy-3D, 2019-08-14:
> Do note that i've had issues with UBL before (it's an IDEX machine) so that might not be a very valid data point. 

Not true!   If IDEX has a problem with UBL...   I'll get it fixed ASAP.   I primarily use IDEX now.   And if there is a sickness I can replicate with UBL...   I assure you, it will get corrected.

orcinus, 2019-08-14:
> Not true! If IDEX has a problem with UBL... I'll get it fixed ASAP. I primarily use IDEX now. And if there is a sickness I can replicate with UBL... I assure you, it will get corrected.

Ah! Apologies, haven't given it a try in quite a while.
Last time i've tried it (months ago) it was a nightmare.

If it's finally fixed, i might ditch MBL then...

InsanityAutomation, 2019-08-15:
@orcinus weve gotten most of it at this point. One thing ive found to kinda ditch the last possible thing is to extend the mesh over the park points so it doesnt cross the boundaries.

orcinus, 2019-08-15:
> @orcinus weve gotten most of it at this point. One thing ive found to kinda ditch the last possible thing is to extend the mesh over the park points so it doesnt cross the boundaries.

That's this, right?
https://github.com/MarlinFirmware/Marlin/issues/10738

InsanityAutomation, 2019-08-15:
Bingo!

orcinus, 2019-09-24:
No clue, i'll give it a try and report back.

boelle, 2019-10-12:
@orcinus any news?

InsanityAutomation, 2019-10-15:
As of a week ago this configuration still had compile errors following the pos struct refactor so its unlikely anyone has been able to test.

boelle, 2019-11-06:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

boelle, 2019-12-01:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-01 by boelle