MarlinFirmware/Marlin #14949: [BUG] Wont properly start print after finished print

TrollNick, 2019-08-14:
### Description

After homing the X-Carriage rams into the edge and the extruder starts extruding vast amounts of filament.
This only happens if I start a print after a print has finished. Canceling a print, in my case via octoprint, and starting a new one without a Hardware reset works fine.
When I finish a print, i have to click the Resetbutton on the board to avoid the issue.

### Steps to Reproduce

1. Finish a print
2. Start a second print

**Expected behavior:**
After homing the nozzle should go down and left just above the printbed and the printer should start printing

**Actual behavior:**
After homing the nozzle goes up and right, to the physical limit of the machine and the extruder starts to extrude vast amounts of filament very quick.

#### Additional Information

Im using a SKR 1.3 Board and TMC5160 for X and Y, DRV8825 for Extruder and Z
I also included octoprints Serial.log, maybe this helps anybody.
Maybe related to #14947

[Conf_and_log.zip](https://github.com/MarlinFirmware/Marlin/files/3501166/Conf_and_log.zip)

2kingsrock, 2019-08-14:
I have had my machine ram into the home or slam into the front of the system when beginning a print.  It seems to always complete the start Gcode (cleaning/prep), but malfunctions when traveling to the edge of the model to print.  I have near identical setup SKR 1.3 CORE XY.  I have tried sticking to DRV8825 for all driver slots to troubleshoot.  My setup is using X, Y, Za (left side), E0 (extruder), E1(Zb right side).  Manual inductance Probe on Zmin.  When this failure occurs, do the values of X, Y, Z on the LCD screen match with where your carriage is, mine do not?  What is your behavior on a bed mesh setup, does it complete properly?   I can pull more logs and continue testing when I have time tonight.  I am also going to run jobs from the LCD SD card to see if this issue occurs.  I can't wait to get this running smoothly.  Some of my errors seem so random.

TrollNick, 2019-08-27:
I only have access to the machine at work, so sorry for the late reply.

Here is what i tested today:
- Started machine
- Finished a print
- Sent G28 via Octoprint
- Moved axis via display. Everything was exactly as expected. Axis moved in the right direction and 
  position on the display was aligned with the real position of the nozzle.
- Started a new print.
- Same faulty behavior occured (Ran into machine limits and extruded like crazy after home went through without a problem)

Made a 2nd run:
- Reset Printer
- Finish a print
- Start the same print again
- Homing worked just fine, Position right after homing was correct
- Then the printer moved to the Machine limits (X_MAX_POS and Y_MAX_POS) and showed these on the Display (X230 Y220 X13.33) So the position of the nozzle aligns with what is shown in the display.

But ofcourse the GCode does not tell the machine to go to X230 Y220. It seems like the machine just goes "You know what i dont care i go to the Limits and just extrude."

boelle, 2019-09-24:
@TrollNick is the problem still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle