MarlinFirmware/Marlin #14952: [BUG] no bed pid autotune in menu, only extruder

sl1pkn07, 2019-08-14:
### Description

Active PIDTEMPBED and PID_AUTOTUNE_MENU only show pids for extruder, but not for bed

### Steps to Reproduce

1. [First Step]
enable PID_AUTOTUNE_MENU
enable PID_EDIT_MENU 
enable PIDTEMPBED
2. [Second Step]
go to LCD -> configuration -> advance -> temperature

**Expected behavior:** [What you expect to happen]

see the pid edit and autotune for bed

**Actual behavior:** [What actually happens]

only show the autotune and edit for extruder

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

https://github.com/sl1pkn07/Marlin/commit/bf7879df42289a81fc08b7f578fa49ee4ec65b14#diff-8270513bcf2a548cd418cdfe123f27d3R462-L540
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

shitcreek, 2019-08-15:
This isn't a bug as it is not in the code. This would be more a feature request. I'm working on getting it added.

kaisertheone, 2020-04-12:
This Feature request is still open. I am missing that option too now.
i flew over the code and found some "leftovers" regarding this point.

`#if ENABLED(PID_AUTOTUNE_MENU)

  #if ENABLED(PIDTEMP)
    int16_t autotune_temp[HOTENDS] = ARRAY_BY_HOTENDS1(150);
  #endif

  #if ENABLED(PIDTEMPBED)
    int16_t autotune_temp_bed = 70;
  #endif

  void _lcd_autotune(const int16_t e) {
    char cmd[30];
    sprintf_P(cmd, PSTR("M303 U1 E%i S%i"), e,
      #if HAS_PID_FOR_BOTH
        e < 0 ? autotune_temp_bed : autotune_temp[e]
      #elif ENABLED(PIDTEMPBED)
        autotune_temp_bed
      #else
        autotune_temp[e]
      #endif
    );
    lcd_enqueue_one_now(cmd);
  }`

unfortunately i cant figure out whats the problem.

sl1pkn07, 2020-07-01:
The FR still up

sjasonsmith, 2020-07-01:
@sl1pkn07 have you tested with `bugfix-2.0.x`, after the change was merged 20 days ago?

sl1pkn07, 2020-07-01:
Oh. First notice . Sorry

Closing

closed 2020-07-01 by sl1pkn07