MarlinFirmware/Marlin #14954: MKS mini Ruramps

Lino77, 2019-08-14:
hello I have a Ruramps4d with DUE and would like to use tmc2130 together with the MKSmini. Unfortunately, I always get this error when compiling

Compiling .pio/build/DUE/src/src/HAL/HAL_DUE/usb/sd_mmc_spi_mem.cpp.o
In file included from Marlin/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi.cpp:65:0:
Marlin/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: variable or field 'u8g_SetPIOutput_DUE' declared void
 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);
                          ^~~~~
Marlin/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi_shared.h:27:26: error: 'u8g_t' was not declared in this scope
Marlin/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi_shared.h:27:26: note: suggested alternative: 'uid_t'
 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);
                          ^~~~~
                          uid_t
Marlin/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi_shared.h:27:33: error: 'u8g' was not declared in this scope
 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);
                                 ^~~
Marlin/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi_shared.h:27:46: error: expected primary-expression before 'pin_index'
 void u8g_SetPIOutput_DUE(u8g_t *u8g, uint8_t pin_index);
                                              ^~~~~~~~~
Marlin/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi_shared.h:28:25: error: variable or field 'u8g_SetPILevel_DUE' declared void
 void u8g_SetPILevel_DUE(u8g_t *u8g, uint8_t pin_index, uint8_t level);
                         ^~~~~
compilation terminated due to -fmax-errors=5.
Compiling .pio/build/DUE/src/src/HAL/HAL_DUE/usb/udc.c.o
*** [.pio/build/DUE/src/src/HAL/HAL_DUE/dogm/u8g_com_HAL_DUE_sw_spi.cpp.o] Error 1

Bob-the-Kuhn, 2019-08-15:
Sorry to be off topic …

I ordered a couple of RURAMPs back in April. but haven't received them yet.  When did you order yours and when did you get it?

Bob-the-Kuhn, 2019-08-15:
I tried a few simple configurations but was unable to reproduce the problem.

Please ZIP & attach the following files:
- Configuration.h
- Configuration_adv.h
- pins_YOUR_BOARD.h

Lino77, 2019-08-15:
@Bob-the-Kuhn I already have the board since January 2018, both have the 1.1 as aich the 1.3 with thermocoler, but never really had time for it
here my config

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3505911/Config.zip)

Bob-the-Kuhn, 2019-08-15:
Strange - the U8G library isn't being included.

Everything is good if I switch to the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

It may take a while to find the root cause.  It's been a while since I've been into the LCD code.

---

I'm curious about the serial port assignments.  

I'm used to setting `#define SERIAL_PORT -1`.  That way the SD card on the LCD is automatically mapped as a drive on the PC.

Why do you have `#define SERIAL_PORT_2 2`?

Bob-the-Kuhn, 2019-08-15:
In the file **\Marlin\src\HAL\HAL_DUE\dogm\u8g_com_HAL_DUE_sw_spi_shared.h** add the line `#include <U8glib.h>` immediately after line 25.

That took care of the compile errors for me.

Lino77, 2019-08-16:
#define SERIAL_PORT_2 2 is for ESP3d wifi modul 

okay it compile but no screen

Lino77, 2019-08-16:
[image]
[image]
the skr 1.3 works perfekt

Bob-the-Kuhn, 2019-08-16:
Good news!

What is a "skr 1.3"?  I gave up trying to remember all the options and printers a long time ago.

Bob-the-Kuhn, 2019-08-16:
Please close this case if everything is working.

Evg33, 2019-08-16:
In my config:

```
#define SERIAL_PORT -1
//-1 - Due Native
// 0 - Due Prog
// 1 - MKS-TFT/NEXTION
// 2 - ESP8266
// 3 - ?

#define SERIAL_PORT_2 1
// 0 - Due Prog
// 1 - MKS-TFT/NEXTION
// 2 - ESP8266

```
All working fine.
RuRamps1.3 + due in zav max

Lino77, 2019-08-16:
BIGTREETECH-SKR-V1.3 is the first picture

Lino77, 2019-08-16:
@EVG witch display you use only the MKS-TFT the nextion did not work or what for a firmwae you use for it

Lino77, 2019-08-16:
if i use REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER) it works fine

Evg33, 2019-08-16:
> @evg witch display you use only the MKS-TFT the nextion did not work or what for a firmwae you use for it

MKS-TFT

Evg33, 2019-08-16:
Sorry.... On photos not mks-tft. 
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
it's works with FULL_GRAPHIC_SMART_CONTROLLER from SKR.

Lino77, 2019-08-16:
if I activate the MMS_mini I also get driver overtemperature warning! on all drivers

boelle, 2019-09-24:
@Lino77 problem solved?

Lino77, 2019-09-28:
No, I think it's on the Ruramps Board that it's not working.

closed 2019-09-28 by Lino77