MarlinFirmware/Marlin #14957: Marlin 2.0 on a geeetech a10 board gt2560 g27 code

brett18, 2019-08-14:
I have been have problems getting the configuration,h to work on.

error: static assertion failed: NOZZLE_PARK_FEATURE is required to use G27 in EVENT_GCODE_SD_STOP.

   static_assert(nullptr == strstr(EVENT_GCODE_SD_STOP, "G27"), "NOZZLE_PARK_FEATURE is required to use G27 in EVENT_GCODE_SD_STOP.");

Error compiling for board Arduino/Genuino Mega or Mega 2560.

the config file I am using is GT2560. 

ANY HElP PLZ 

many thanks 
Brett

thisiskeithb, 2019-08-15:
As of the latest commit (https://github.com/MarlinFirmware/Marlin/commit/179d6c4ed115e67ae826b5738e0009c6be2d58ea), the [default A10 config](https://github.com/MarlinFirmware/Marlin/tree/179d6c4ed115e67ae826b5738e0009c6be2d58ea/config/examples/Geeetech/A10) builds without any errors.

Have you modified your copy?

zonalimitatore, 2019-08-19:
What version of the board?

brett18, 2019-08-20:
Thank you for replying. It's a v3 board. I sat down at the week end and have all most got it working 100%. The only thing left to put right is to turn the Extruder stepper to false and set up my z offset in relationship to the bl touch. Then hopefully it's all working fine. And it will just be tuning the steeper and feeds for my printer.

zonalimitatore, 2019-09-24:
@Brett18
Do you notice a stop in X0 Y225 (angle front left) when you start a bed levelling using Marlin program for Windows?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle