MarlinFirmware/Marlin #14958: Marlin 2.0  Homing Z won't work with SKR Pro

budi67, 2019-08-15:
Hello all

 I just finished install Marlin 2.0 on my SKR PRO, but i have a problem with homing.

I have a bltouch and i must set my Z_offset to -8 
Bltouch ==> https://www.thingiverse.com/thing:2841443

But when it homing it stop at z0, i don't understant

A big thanks
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3506200/Configuration.zip)

golfromeo-fr, 2019-08-15:
with the updates in the last 48 hours, the bltouch probing with my SKR PRO no longer works.
the bltouch now deploy/stow 3 times and no longer go down for probing
the probing fails

golfromeo-fr, 2019-08-15:
is it related to 
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/31
?

christianger, 2019-08-15:
> with the updates in the last 48 hours, the bltouch probing with my SKR PRO no longer works.
the bltouch now deploy/stow 3 times and no longer go down for probing
the probing fails

This sound like the same issue i have.

golfromeo-fr, 2019-08-15:
I have tried the M43 S command like @christianger  did

I have tried 
#define Z_MIN_PROBE_ENDSTOP_INVERTING false
and 
#define Z_MIN_PROBE_ENDSTOP_INVERTING true

but the sequence only prints, ignoring the Z_MIN_PROBE_ENDSTOP_INVERTING value
Recv: . Z_MIN_PROBE_ENDSTOP_INVERTING: false

Send: M43 S
Recv: Servo probe test
Recv: . using index:  0, deploy angle: 10, stow angle:   90
Recv: . Probe Z_MIN_PROBE_PIN: 11
Recv: . Z_MIN_PROBE_ENDSTOP_INVERTING: false
Recv: . Check for BLTOUCH
Recv: . Deploy & stow 4 times
Recv: WARNING: INVERTING setting probably backwards.
Recv: ** Please trigger probe within 30 sec **
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv: FAIL: No trigger detected
Recv: ok

budi67, 2019-08-16:
Hello

How do you active M43 ?
I have try to active #define PINS_DEBUGGING, but i have this error !!
Thanks
[image]

golfromeo-fr, 2019-08-16:
> I have try to active #define PINS_DEBUGGING, but i have this error !!

https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/31#issuecomment-521911559

Patag, 2019-08-16:
@budi67  See #14894

budi67, 2019-08-16:
Thanks Patag.

I have download a nex firmware from Marlin and all started again from 0.
All work fine.

But as the SKR Pro as no Eeprom, where save the Mesh ?

christianger, 2019-08-16:
You can use SD-Card EEPROM Emulation to store the values in the SD-Card of the LCD:

#define BOARD_NAME "BIGTREE SKR Pro 1.1"

// Use one of these or SDCard-based Emulation will be used
//#define SRAM_EEPROM_EMULATION   // Use BackSRAM-based EEPROM emulation
//#define FLASH_EEPROM_EMULATION  // Use Flash-based EEPROM emulation        

simply uncomment FLASH_EEPROM_EMULATION

(Marlin/src/pins/stm32/pins_BIGTREE_SKR_PRO_V1.1.h)

budi67, 2019-08-16:
A big thanks for your answer Christianger, i will try this

budi67, 2019-08-16:
I have uncomment   FLASH_EEPROM_EMULATION

For save it's M500 ??
I have try and i get this : 
Error:EEPROM disabled
[ERROR] Error:EEPROM disabled

gloomyandy, 2019-08-16:
@budi67 Do you have EEPROM_SETTINGS enabled? That needs to be on toy have M500 etc work.

budi67, 2019-08-16:
Thanks Gloomyandy, it's ok, work fine

budi67, 2019-08-16:
Sorry but another question about start gcode in slicer

G28 after just M420 S1 ?

budi67, 2019-08-16:
In prontface when i make M500 after a G29
It say : echo:Settings Stored (762 bytes; crc 20618)
If i make M501 : echo:EEPROM version mismatch (EEPROM=? Marlin=V68)
echo:Hardcoded Default Settings Loaded

boelle, 2019-09-24:
@budi67 problem solved?

budi67, 2019-09-24:
yes, problem solved, sorry for not given assets news, lots of worries right now

closed 2019-09-24 by budi67