MarlinFirmware/Marlin #14962: [QUESTION] How does Marlin handle non-English languages?

marcio-ao, 2019-08-15:
Is there any documentation on how Marlin handles non-English languages? One of the issues with the UI we developed for our TAZ Pro is that it currently has no localization features. In fact, it is limited to ASCII, which doesn't even allow us to do accents for European languages, not to mention any of the other languages. I would like to explore ways to support localization in our UI, but I would rather build upon what Marlin already has, than re-inventing everything from scratch.

Is there an explanation anywhere on how Marlin handles this? A few questions I need answered:

- How are extended characters encoded in Marlin (is it unicode?)
- What format are Marlin's current fonts in?
- What is the process for converting encoded characters into glyph indexes in a particular bitmap font?

One pie-in-the-sky idea I had was compiling FreeType for the SAM8X, but my guess is that there won't be enough RAM for rendering fonts on-the-fly, so I may have to make do with a bitmap font. I doubt Marlin has fonts with a resolution suitable to our 800x480 display, so my guess is that I would have to substitute a much higher resolution font, but if I could do so in a way that still worked with the localization strings in Marlin, that would be ideal.

RudyBenner, 2019-08-15:
http://marlinfw.org/docs/development/lcd_language.html

igorepi, 2019-08-16:
If you want to show file names on your display there is one more question - 
  How does Marlin handle non-English file names ?

And the answer is - currently Marlin does not handle such file names at all. Long file names in FAT/FAT32 filesystems are stored in UTF-16 encoding; Marlin in SdBaseFile.cpp just discards high byte of long file name encoding.

psavva, 2019-08-21:
It used the internal u8glib font format.
see https://github.com/olikraus/u8glib/wiki/bdf2u8g how to convert a font file to the u8glib format.

Then it's a car of mapper as done here:
https://github.com/thinkyhead/Marlin/blob/928ff6487ad88dc05c3039b53a30e7f6a15f609a/Marlin/utf_mapper.h

This was the PR for adding Greek which also required a new font and utf8 mapper.
https://github.com/MarlinFirmware/Marlin/pull/4024/files

AnHardt, 2019-08-22:
My old font system is obsolete.
Better have a look into https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/buildroot/share/fonts/uxggenpages.md

boelle, 2019-09-24:
@marcio-ao figured this one out?

marcio-ao, 2019-09-24:
Yes. I'll close this ticket.

closed 2019-09-24 by marcio-ao