MarlinFirmware/Marlin #14963: Bilinear Subdivision Unwanted Z movement

blueeagle69, 2019-08-15:
-->

### Description With Bilinear  Subdivion bug? Not quite sure it's the same as #10730

 During the print, the first layer goes down fine, but when the nozzle reaches the right hand side, I get get extreme Z  movement (up) beyond the level compensation. Then as it moves towards the back, the Z moves down again, and carries on laying down material. https://www.youtube.com/watch?v=j_xGobMk9ZQ&t=12s

### Steps to Reproduce Enable Bed leveling (bilinear) and subdivision.

1. [First Step] Run Bilinear Bed Levelling with Bilinear Subdivision enabled with Extrapolate Beyond Grid.
2. [Second Step] Start print.
3. [and so on...]

**Expected behavior:** [What you expect to happen] Z should carry on compensating for the bed tilt

**Actual behavior:** [What actually happens] This works fine up until it hits the far right of the bed

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
I have attached my config files. My copy of Marlin 2.0 is around 2 weeks old.
Board: SKR 1.3 with Bltouch 3.1
Apologies if this is indded the same bug as #10730
I still have Extrapolate Beyond Grid enabled.
Amazingly I never noticed this issue until I printed a larger object, and then for some reason Subdivision and Extrapolate Beyond Grid popped into my strange head, so I managed to sort the problem quite quickly by disabling Subdivision so all is good!
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3507395/Configuration.zip)

golfromeo-fr, 2019-08-16:
I am using the EXTRAPOLATE_BEYOND_GRID and ABL_BILINEAR_SUBDIVISION
I have just observed the behavior
at first I have thought what is cura 4.2.1 doing: I have a raft at 0.3mm the extruder goes at the right  (border) of the raft, the Z axis goes up for 25 mm and drop filament (raft) then come back to 0.3mm!!!
I guess now it is not cura but this feature ABL_BILINEAR_SUBDIVISION :)

#if ENABLED(AUTO_BED_LEVELING_BILINEAR)

    // Beyond the probed grid, continue the implied tilt?
    // Default is to maintain the height of the nearest edge.
    #define EXTRAPOLATE_BEYOND_GRID

    //
    // Experimental Subdivision of the grid by Catmull-Rom method.
    // Synthesizes intermediate points to produce a more detailed mesh.
    //
    #define ABL_BILINEAR_SUBDIVISION
    #if ENABLED(ABL_BILINEAR_SUBDIVISION)
      // Number of subdivisions between probe points
      #define BILINEAR_SUBDIVISIONS 2
    #endif

  #endif

edwilliams16, 2019-09-05:
@virtualight  has been trying out Catmull-Rom subdivision and observes a perhaps related bug:
Note that part of the last (33) column of interpolated data is missing - shown as =======
There's no reason it should be, I calculate the missing column. It should be:
[2.627,2.54919,2.47137,2.41,2.37741,2.36126,2.341,2.31726,2.28941,2.235,
2.13385,2.00615,1.89,1.79411,1.70978,1.649,1.61793,1.61041,1.62,1.65185,
1.70081,1.745,1.77541,1.80104,1.827,1.8577,1.88874,1.912,1.92267,1.92556,1.927,
1.92863,1.92881,1.929]
The [0,3,6...] points are sample data - the points in between Catmull-Rom interpolated.
If that last column has bad/missing/corrupted whatever data, it might explain the problem noted in this thread.

READ:       0      1      2      3      4      5      6      7      8      9     10     11
READ:  0 +1.659 +1.745 +1.784 +1.849 +1.920 +1.969 +2.084 +2.194 +2.320 +2.424 +2.514 +2.627
READ:  1 +1.551 +1.731 +1.814 +1.843 +1.898 +1.949 +2.000 +2.059 +2.124 +2.200 +2.294 +2.410
READ:  2 +1.541 +1.631 +1.733 +1.765 +1.800 +1.875 +1.894 +1.867 +1.886 +1.996 +2.163 +2.341
READ:  3 +1.502 +1.569 +1.592 +1.553 +1.578 +1.653 +1.659 +1.573 +1.588 +1.765 +1.959 +2.235
READ:  4 +1.382 +1.371 +1.351 +1.271 +1.212 +1.212 +1.186 +1.155 +1.180 +1.286 +1.500 +1.890
READ:  5 +1.304 +1.204 +1.153 +1.053 +0.975 +0.933 +0.920 +0.898 +0.924 +1.004 +1.200 +1.649
READ:  6 +1.365 +1.220 +1.139 +1.024 +0.969 +0.943 +0.949 +0.920 +0.924 +0.994 +1.208 +1.620
READ:  7 +1.512 +1.392 +1.292 +1.157 +1.094 +1.094 +1.133 +1.159 +1.169 +1.243 +1.394 +1.745
READ:  8 +1.669 +1.629 +1.549 +1.427 +1.341 +1.327 +1.392 +1.476 +1.506 +1.537 +1.596 +1.827
READ:  9 +1.802 +1.837 +1.825 +1.680 +1.559 +1.569 +1.637 +1.763 +1.829 +1.804 +1.800 +1.912
READ: 10 +1.878 +1.898 +1.882 +1.818 +1.778 +1.788 +1.853 +1.933 +1.982 +1.924 +1.859 +1.927
READ: 11 +2.080 +2.080 +2.096 +2.106 +2.090 +2.098 +2.173 +2.204 +2.198 +2.098 +1.963 +1.929
READ: 
READ: Subdivided with CATMULL ROM Leveling Grid:
READ:         0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15       16       17       18       19       20       21       22       23       24       25       26       27       28       29       30       31       32       33
READ:  0 +1.65882 +1.68932 +1.71983 +1.74510 +1.76071 +1.77110 +1.78431 +1.80378 +1.82607 +1.84902 +1.87291 +1.89746 +1.91961 +1.93508 +1.94815 +1.96863 +2.00247 +2.04372 +2.08431 +2.12077 +2.15657 +2.19412 +2.23559 +2.27880 +2.31961 +2.35635 +2.39070 +2.42353 +2.45374 +2.48243 +2.51373 +2.54989 +2.58867  =======
READ:  1 +1.61924 +1.66333 +1.70741 +1.74372 +1.76717 +1.78284 +1.79818 +1.81507 +1.83163 +1.84975 +1.87116 +1.89413 +1.91525 +1.93114 +1.94519 +1.96413 +1.99178 +2.02433 +2.05701 +2.08790 +2.11893 +2.15113 +2.18568 +2.22140 +2.25577 +2.28764 +2.31817 +2.34829 +2.37735 +2.40602 +2.43726 +2.47292 +2.51115  =======
READ:  2 +1.57967 +1.63733 +1.69500 +1.74234 +1.77362 +1.79458 +1.81206 +1.82636 +1.83720 +1.85047 +1.86942 +1.89080 +1.91089 +1.92721 +1.94223 +1.95962 +1.98109 +2.00493 +2.02970 +2.05503 +2.08128 +2.10813 +2.13577 +2.16400 +2.19194 +2.21893 +2.24563 +2.27306 +2.30097 +2.32961 +2.36078 +2.39594 +2.43363  =======
READ:  3 +1.55098 +1.61474 +1.67850 +1.73137 +1.76805 +1.79383 +1.81373 +1.82651 +1.83341 +1.84314 +1.85970 +1.87909 +1.89804 +1.91532 +1.93217 +1.94902 +1.96572 +1.98243 +2.00000 +2.01881 +2.03849 +2.05882 +2.07952 +2.10087 +2.12353 +2.14749 +2.17277 +2.20000 +2.22927 +2.26049 +2.29412 +2.33108 +2.37044  =======
READ:  4 +1.54154 +1.59787 +1.65419 +1.70305 +1.74183 +1.77314 +1.79731 +1.81139 +1.81833 +1.82731 +1.84168 +1.85810 +1.87560 +1.89501 +1.91550 +1.93370 +1.94791 +1.95982 +1.97110 +1.98151 +1.99128 +2.00276 +2.01600 +2.03093 +2.04975 +2.07322 +2.10057 +2.13159 +2.16685 +2.20578 +2.24648 +2.28890 +2.33308  =======
READ:  5 +1.54299 +1.58440 +1.62580 +1.66514 +1.70359 +1.73997 +1.76870 +1.78513 +1.79391 +1.80341 +1.81570 +1.82872 +1.84466 +1.86673 +1.89173 +1.91227 +1.92542 +1.93410 +1.93980 +1.94086 +1.93893 +1.94038 +1.94613 +1.95526 +1.97139 +1.99622 +2.02805 +2.06536 +2.10912 +2.15836 +2.20864 +2.25883 +2.31006  =======
READ:  6 +1.54118 +1.57081 +1.60044 +1.63137 +1.66710 +1.70414 +1.73333 +1.74887 +1.75657 +1.76471 +1.77473 +1.78519 +1.80000 +1.82397 +1.85229 +1.87451 +1.88686 +1.89310 +1.89412 +1.88671 +1.87407 +1.86667 +1.86638 +1.87131 +1.88627 +1.91409 +1.95192 +1.99608 +2.04699 +2.10421 +2.16275 +2.22135 +2.28126  =======
READ:  7 +1.53326 +1.55966 +1.58605 +1.61271 +1.64237 +1.67230 +1.69448 +1.70287 +1.70352 +1.70661 +1.71467 +1.72517 +1.74067 +1.76607 +1.79647 +1.81968 +1.83177 +1.83667 +1.83406 +1.81878 +1.79599 +1.78076 +1.77619 +1.77918 +1.79550 +1.82984 +1.87750 +1.93014 +1.98558 +2.04600 +2.10959 +2.17678 +2.24714  =======
READ:  8 +1.52208 +1.54839 +1.57470 +1.59818 +1.61940 +1.63780 +1.64887 +1.64688 +1.63756 +1.63370 +1.63962 +1.65101 +1.66761 +1.69368 +1.72497 +1.74851 +1.76061 +1.76496 +1.75962 +1.73736 +1.70541 +1.68352 +1.67612 +1.67878 +1.69797 +1.74045 +1.79946 +1.86115 +1.91977 +1.98106 +2.04837 +2.12480 +2.20724  =======
READ:  9 +1.50196 +1.52578 +1.54960 +1.56863 +1.58199 +1.59056 +1.59216 +1.58134 +1.56354 +1.55294 +1.55483 +1.56391 +1.57843 +1.60218 +1.63137 +1.65294 +1.66340 +1.66623 +1.65882 +1.63312 +1.59717 +1.57255 +1.56427 +1.56732 +1.58824 +1.63450 +1.69862 +1.76471 +1.82505 +1.88736 +1.95882 +2.04488 +2.14009  =======
[... cut ...]
READ: 32 +2.00842 +2.01075 +2.01308 +2.01511 +2.01731 +2.01921 +2.01903 +2.01572 +2.01033 +2.00421 +1.99647 +1.98800 +1.98286 +1.98120 +1.98286 +1.99143 +2.01125 +2.03798 +2.06216 +2.08107 +2.09744 +2.11002 +2.12084 +2.12787 +2.12382 +2.10349 +2.07208 +2.03784 +1.99835 +1.95605 +1.92643 +1.91892 +1.92409 +1.92927
READ: 33 +2.08039 +2.07981 +2.07923 +2.08039 +2.08468 +2.09070 +2.09608 +2.10073 +2.10472 +2.10588 +2.10167 +2.09463 +2.09020 +2.08860 +2.08962 +2.09804 +2.11954 +2.14844 +2.17255 +2.18758 +2.19782 +2.20392 +2.20821 +2.20835 +2.19804 +2.17299 +2.13747 +2.09804 +2.05178 +2.00160 +1.96275 +1.94408 +1.93675 +1.92941

edwilliams16, 2019-09-06:
More data:
With GRID_MAX_POINTS_X 13   z(36,0:20)  are undefined
With GRID_MAX_POINTS_X 12  z(33,0:17) are undefined
With GRID_MAX_POINTS_X 10  none are undefined.

boelle, 2019-09-24:
@blueeagle69 is this still a problem?

edwilliams16, 2019-09-24:
@boelle   Unless there's been a recent bugfix, using GRID_MAX_POINTS_X >10 results in NaN's in the
z_values_virt array.

blueeagle69, 2019-09-25:
@boelle I will have to re enable sub division and test. Cheers

blueeagle69, 2019-10-04:
@boelle Really sorry for the delay. I can confirm it's still a problem.

boelle, 2019-10-21:
@Roxy-3D can you shine a bit of light on this one? it should be possible to go to 15 but it seems not to be the case on this one

boelle, 2019-11-03:
@blueeagle69 still an issue with all the updates that have been done?

blueeagle69, 2019-11-04:
@belle I will try enabling Sub Division tonight and report back. I flashed another new build last night.

blueeagle69, 2019-11-04:
@boelle 

I can confirm that the bug with Bilinear Subvision  has appeared to have been fixed :)
Thanks for reminding me.
I disabled the feature and carried on.
Last night I compiled and flashed the latest build that carries updates for LPC176x chipsets, so I haven't got a clue when it got fixed.

Thanks again 😀
Shall I close this?

closed 2019-11-04 by boelle