MarlinFirmware/Marlin #14964: Nozzle and Bed temperature shows always a high value

erlsen, 2019-08-15:
hey guys, i got a problem with my CR 10S Pro

bed + nozzle temperature shows always(even if hes cold about 155 degrees nozzle temp and also a high bed temp.)

ive update 2 days ago the firm and screenware. (CR10SPro_B7_Stock_115k)
[image]

if i reinstall the original firmware the error still there

shitcreek, 2019-08-16:
Do you have the correct thermistor type selected? Have you checked the physical connections?
We would need more information such as the configuration to have a better clue.

erlsen, 2019-08-16:
ive checked the physical connecttions, and also changed to a new thermistor. same result

how do i chose the correct thermistor type?
what kind of configration do u mean and need?

shitcreek, 2019-08-16:
Marlin configuration files, particularly configuration.h and configuration_adv.h.
You need to make sure the correct type of sensor is set here:
```
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_BED 5
#define TEMP_SENSOR_CHAMBER 0
```
- this is found inside of configuration.h

If you haven't gotten this far, then you've yet to determine this to be a bug or operator error. This place is reserved for bug reporting and feature requests. Next time please try these other sources first:
- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-09-08 by shitcreek