MarlinFirmware/Marlin #14968: TMC2208 Serial 1cable support  LPC1768

lightface79, 2019-08-16:
Hi   i saw a nice little project on a LPC1768 and 1 cable Serial hookup without eany 1K resistors
And just using 1 pin / each Driver (TMC2208)

The link for the project   https://github.com/MarlinFirmware/Marlin/issues/13197

It´s closed now so i cant ask eany Q to the authers.

Is this "Alpa" system in place in the latest marlin 2.0 Betas??

I was running klipper for a nother project and in klipper they used this system and it worked like a charm just using 1 cable and 1 pin on the controler For both RX/tx to the tmc2208

All info is gratly apriciated.

Best regards Janne // Sweden

shitcreek, 2019-08-16:
That thread is closed but you can still comment there and you can also message the author directly.
@Msq001

lightface79, 2019-08-16:
Thanks Shitcreek

lightface79, 2019-08-16:
downt know hot to MSG users =(       but still... want to know if eanything of this is implemented in the recent bulids =)

gloomyandy, 2019-08-16:
Yes the half duplex code (single wire implementation) is in the current LPC176x framework, so available for use.

boelle, 2019-09-24:
will close this one

@lightface79 you can always ask on discord

-Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-09-24 by boelle