MarlinFirmware/Marlin #14973: [BUG] Temperature Updates After Every Line

ajaybnl, 2019-08-17:
why is the Temperature is coming ?
14:07:09.595 : ok
14:07:09.595 : N1592 G1 F1800 X36.589 Y108.810 E2718.0790552
14:07:09.610 : ok
14:07:09.610 : N1593 G0 X36.589 Y108.27 F6000123
14:07:10.268 : T:200.09 /200.00 B:36.81 /0.00 @:58 B@:0
14:07:11.274 : T:199.93 /200.00 B:36.77 /0.00 @:63 B@:0
14:07:12.284 : T:200.18 /200.00 B:36.77 /0.00 @:56 B@:0
14:07:13.292 : T:200.09 /200.00 B:36.75 /0.00 @:59 B@:0
14:07:13.672 : ok
14:07:13.672 : N1594 M117 ETE 4h 09m 17s4
14:07:13.681 : ok
14:07:13.681 : N1595 G1 F1800 X156.769 Y108.270 E2725.610402
14:07:13.693 : ok
14:07:13.693 : N1596 G0 X156.769 Y107.73 F600075
14:07:14.303 : T:200.04 /200.00 B:36.77 /0.00 @:60 B@:0
14:07:15.308 : T:199.93 /200.00 B:36.81 /0.00 @:63 B@:0
14:07:16.315 : T:199.93 /200.00 B:36.83 /0.00 @:62 B@:0
14:07:17.324 : T:200.09 /200.00 B:36.83 /0.00 @:58 B@:0
14:07:17.754 : ok
14:07:17.754 : N1597 G1 F1800 X36.589 Y107.730 E2733.1417555
14:07:17.775 : ok
14:07:17.775 : N1598 G0 X36.589 Y107.19 F6000114
14:07:18.349 : T:199.97 /200.00 B:36.81 /0.00 @:62 B@:0
14:07:19.357 : T:199.97 /200.00 B:36.85 /0.00 @:62 B@:0
14:07:20.364 : T:200.00 /200.00 B:36.81 /0.00 @:61 B@:0
14:07:21.371 : T:199.93 /200.00 B:36.83 /0.00 @:62 B@:0
14:07:21.839 : ok
14:07:21.839 : N1599 M117 ETE 4h 09m 09s6
14:07:21.845 : ok
14:07:21.845 : N1600 G1 F1800 X156.769 Y107.190 E2740.6731012
14:07:21.860 : ok
14:07:21.860 : N1601 G0 X156.769 Y106.65 F600064
14:07:22.382 : T:200.00 /200.00 B:36.90 /0.00 @:60 B@:0

see?

golfromeo-fr, 2019-08-17:
if you are using, say Octoprint, it is used as a way to keep the communication channel alive

AnHardt, 2019-08-17:
Because of
```
#define AUTO_REPORT_TEMPERATURES
```
in `Configuration_adv.h`.

closed 2019-08-19 by ajaybnl