MarlinFirmware/Marlin #14974: [BUG] LPC1768 I2C display doesn't work

CReimer, 2019-08-17:
### Description

I'm trying to connect a 2004 LCD with a PCF8574AT chip on my newly bought Bigtreetech SKR V1.3. There's no dedicated I2C header on this board, so I had to find a more creative solution. SCL0 (P0_28) is available on the EXP1 Port and SDA0 (P0_27) is connected to SD_DETECT. A bit of soldering later and I have an I2C available. After that I removed all instances of these two pins in pins_BIGTREE_SKR_V1.3.h

This is the way I set things up in Configuration.h (tested on Mega2560)
```
#define ULTRA_LCD
#define LCD_I2C_ADDRESS 0x3F
#define LCD_I2C_TYPE_PCF8575
#define LCD_WIDTH 20
#define LCD_HEIGHT 4
```

Build fails with this
```
In file included from Marlin/src/lcd/HD44780/lcdprint_hd44780.cpp:22:
Marlin/src/lcd/HD44780/ultralcd_HD44780.h:56:12: fatal error: LCD.h: No such file or directory
```

Solution for this error is editing platformio.ini
```
lib_compat_mode   = soft
```

New error
```
In file included from Marlin/src/lcd/HD44780/ultralcd_HD44780.h:56,
                 from Marlin/src/lcd/HD44780/lcdprint_hd44780.cpp:22:
.pio/libdeps/LPC1768/LiquidCrystal_ID136/LCD.h:53:10: fatal error: WProgram.h: No such file or directory
```

Solution
```
#define ARDUINO 20000
```
in LCD.h, FastIO.h, I2CIO.cpp, LCD.cpp, LiquidCrystal_I2C.cpp, LiquidCrystal_I2C_ByVac.cpp, LiquidCrystal.cpp, LiquidCrystal_SR.cpp and LiquidCrystal_SR3W.cpp

and I had to comment out "pins_arduino.h" in FastIO.h

Additionally I edited Wire.cpp
```
#define USEDI2CDEV_M 0
```
To enable the right I2C

Now Marlin compiles through. But it doesn't start up on the SKR board and I'm out of ideas now.

Any ideas what I did wrong? And is it normal that I had to edit all these files just to get it to compile?

Thanks

gloomyandy, 2019-08-17:
What exactly did you do to make P0_27 available? It is normally the SD detect pin for the onboard SD card reader and has a pullup to 3.3V are you sure you have disconnected it from both of these.

As to all of the rest of the changes, who knows. I think the basic answer is that i2c is not currently supported on this board. Wouldn't it have been easier to connect the 2004 display directly and sort out the pins for it?

@p3p is i2c supported on the LPC176x devices?

CReimer, 2019-08-17:
I unsoldered R38 and connected a jumper wire directly to the pad next to the sd card slot. After that I put a piece of shrinking tube around the actual sd detect lever on the sd card slot to make sure it doesn't short P0_27 to ground.

Checked everything with my multimeter and I do end up on the LPC right next to P0_28

CReimer, 2019-08-17:
Yes, it would have been easier to just connect the 2004 display. But I need most of the pins on EXP1 and EXP2 for an additional Stepper driver and 4 additional fan connectors.

(Sorry for double posting, forgot to answer the second paragraph)

CReimer, 2019-09-24:
Nope. I'm using a different board now

boelle, 2019-10-18:
will close this one as there is no way  of testing it without someone has the hardware

we can always reopen

closed 2019-10-18 by boelle