MarlinFirmware/Marlin #14975: Setting fan speed by LCD requires 4 knob clicks per percent speed

LMF5000, 2019-08-17:
### Description

Running SKR E3 on CR-10 mini stock LCD. Encoder (knob) works fine in all menu items, one click moves one menu item, and for example when setting temperature one click changes the value by 1°C as expected. However when setting fan speed under Temperature -> Fan, it takes four encoder clicks to change speed by 1%.

### Steps to Reproduce

1. Open Temperature -> Fan speed from LCD
2. Try increasing fan speed by 1%

**Expected behavior:**
Rotating the knob one "click" increases fan speed by 1%

**Actual behavior:**
The knob must be turned until it clicks 2-3 times to increment speed by 1%. Sometimes it takes 2 clicks, sometimes it takes 3, sometimes 4 clicks.

#### Additional Information
The other menu items (like nozzle temperature and bed temperature) are working as expected (one knob click per value) so this doesn't seem to be a global setting, whatever it is just affects the fan setting (and possibly other things I haven't stumbled upon yet)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3511971/Configuration.zip)

.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

AnHardt, 2019-08-17:
Formerly the fan value was in 0-127 or 0-255 - i don't really remember.
Someone *improved* that to **show** % recently.
Probably untested this was merged.
I guess the **increasing/decreasing** is still in the old scale. Only the display is in **%**.

Ludy87, 2019-08-17:
Change from 0-255 to 0% -100% was in March
https://github.com/MarlinFirmware/Marlin/pull/13519

I do not suppose that it is because of this change, my guess would be to manipulate the ENCODER settings of the last days

AnHardt, 2019-08-17:
I'd look exactly here.
```diff
-      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(uint8, MSG_FAN_SPEED FAN_SPEED_1_SUFFIX, &thermalManager.lcd_tmpfan_speed[0], 0, 255, thermalManager.lcd_setFanSpeed0);
+      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(percent, MSG_FAN_SPEED FAN_SPEED_1_SUFFIX, &thermalManager.lcd_tmpfan_speed[0], 0, 255, thermalManager.lcd_setFanSpeed0);
```

marcio-ao, 2019-08-19:
> Someone improved that to show % recently.
> Probably untested this was merged.

This was me. For an end-user which isn't a programmer, a percentage makes a lot more sense than 255.

I understand that now one encoder click does not increment the number by one, which is true; however I would still consider the new behavior to be an improvement over the old. If there is disagreement on the merits of the trade-off, perhaps this could be made into a configurable option.

As for actually fixing it so it is both a percentage **and** the encoder wheel increments by one, I suspect this would be a very difficult thing to pull off. Either:

1. Marlin needs to be modified so fan speed are stored internally as a value from 0 to 100.
2. Or the LCD encoder multiplier needs to be changed for that particular menu item only.

Option one will probably require changes many places in Marlin, and will cause problems with M106. Option two would require the ultralcd menu code to be modified for this specific case.

Personally, I think both of these options add complexity to the code that probably ought not to be there. The current behavior is an acceptable tradeoff, in my opinion.

marcio-ao, 2019-08-20:
> Personally, I think both of these options add complexity to the code that probably ought not to be there. The current behavior is an acceptable tradeoff, in my opinion.

Okay. I'll eat my words. Turns out the fix was trivial once I looked at the code more closely. There already is a facility for scaling based on particular data types.

closed 2019-08-21 by thinkyhead

LMF5000, 2019-08-21:
Awesome! Thanks all for getting it fixed :)