MarlinFirmware/Marlin #14977: [BUG] Error: Probing failed. UBL probing same spot repeatedly

MarkJB, 2019-08-17:
### Description

When attempting to setup UBL, building a cold mesh (from the LCD or with G29 P1 (with or without G28, and with various mesh inset values)) homes X, Y & Z then says probing 1/n points but then probes the same spot repeatedly.

With 'levelling debug' enabled I see lots of 'Error:Probling Failed' messages. (see serial output in attached zip)

This is with Bugfix-2.0.x updated to the latest on a BOARD_BIGTREE_SKR_V1_3.

### Steps to Reproduce

1. Configure and enable UBL
2. Compile and upload
3. Reset eeprom (M502 then M500)
4. Enable debug
5. Run G29 P1

**Expected behavior:** [What you expect to happen]

Bed is probed n times in a grid.

**Actual behavior:** [What actually happens]

Bed is probed n times in the same spot.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[ubl_issue.zip](https://github.com/MarlinFirmware/Marlin/files/3512072/ubl_issue.zip)
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

MarkJB, 2019-08-17:
More details of what I've tried here: https://reprap.org/forum/read.php?415,857958,858125

Roxy-3D, 2019-08-17:
Usually....  It is helpful to turn on 
```
//#define DEBUG_LEVELING_FEATURE
```
and then do a M111 S255

After doing that...  Marlin will be very verbose about what is happening at each step of the process.   If you do that and paste the log here...  We should be able to figure out what the problem is.

MarkJB, 2019-08-17:
Debug was enabled, output is in the attached zip but can paste it into a comment or maybe link to a gist if preferred.

MarkJB, 2019-08-17:
Debug output: https://gist.github.com/MarkJB/2fc41edb45e6ac25bd2b058713416c0b

2kingsrock, 2019-08-17:
I see similar issues.  I am running a COREXY with BOARD_BIGTREE_SKR_V1_3 DRV8825.  Fixed Probe.  What I see is the Y does not advance, the X seems to run right to left, then no advance in Y.  So in my quick test, X advanced 3 probe points, then returned back to previous spot.  I believe this should have advanced in Y and continued the probe across X, but Y did not advance.  As strange as this is, the LCD is actually showing values advancing for Y, but the system does not physically advance.  
Also no longer seems to respect the offset for the mesh.  Meaning it will over drive beyond 0, 0 when probing.  I pulled down the latest from this am, still present.  I am using Z2 (dual Z and E1) in my configuration.  I will turn on Debug and pull a log also if that would be helpful.

MarkJB, 2019-08-17:
With `#define UBL_DEVEL_DEBUGGING` enabled, G29 W gives me the following:

> [16:01:32:696] echo:Hardcoded Default Settings Loaded␊
[16:01:32:697] ok␊
[16:01:36:762] echo:Settings Stored (614 bytes; crc 15129)␊
[16:01:36:779] ok␊
[16:01:40:781] echo:DEBUG:INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING␊
[16:01:40:782] ok␊
[16:01:57:808] Unified Bed Leveling System v1.01 inactive␊
[16:01:57:852] No Mesh Loaded.␊
[16:01:57:904] planner.z_fade_height : 0.0000␊
[16:01:57:910] # of samples: 9␊
[16:01:57:910] Mean Mesh Height: 0.000000␊
[16:01:57:910] Standard Deviation: 0.000000␊
[16:01:57:910] zprobe_zoffset: -5.6999998␊
[16:01:57:910] MESH_MIN_X  (_MAX((((200) / 2) - (200) / 2) + 25, -35))=25␊
[16:01:57:954] MESH_MIN_Y  (_MAX((((200) / 2) - (200) / 2) + 25, -2))=25␊
[16:01:58:006] MESH_MAX_X  (_MIN((((200) / 2) + (200) / 2) - (25), 200))=175␊
[16:01:58:056] MESH_MAX_Y  (_MIN((((200) / 2) + (200) / 2) - (25), 200))=175␊
[16:01:58:107] GRID_MAX_POINTS_X  3␊
[16:01:58:159] GRID_MAX_POINTS_Y  3␊
[16:01:58:209] MESH_X_DIST  75.00␊
[16:01:58:211] MESH_Y_DIST  75.00␊
[16:01:58:260] X-Axis Mesh Points at: 25.000  100.000  175.000  ␊
[16:01:58:336] Y-Axis Mesh Points at: 25.000  100.000  175.000  ␊
[16:01:58:413] ␊
[16:01:58:463] ubl_state_at_invocation :0␊
[16:01:58:464] ubl_state_recursion_chk :0␊
[16:01:58:515] Meshes go from 0x000002E8 to 0x00000F7F␊
[16:01:58:566] sizeof(ubl) :  1␊
[16:01:58:566] ␊
[16:01:58:578] z_value[][] size: 36␊
[16:01:58:578] ␊
[16:01:58:591] EEPROM free for UBL: 0x00000C97␊
[16:01:58:643] EEPROM can hold 89 meshes.␊
[16:01:58:644] ␊
[16:01:58:667] Unified Bed Leveling sanity checks passed.␊
[16:01:58:679] ok␊

2kingsrock, 2019-08-17:
I see this in the debug, Recv: >>> do_blocking_move_to(-483.75, -498.75, 5.00)   Recv: >>> do_homing_move(X, -483.75, [50.00])

2kingsrock, 2019-08-17:
[skr_v1.zip](https://github.com/MarlinFirmware/Marlin/files/3512140/skr_v1.zip)

Recv: >>> G28
Recv: Machine Type: Core
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X:45 Y:15 Z:2.00 (Right-Back & Above Nozzle)
Recv: Auto Bed Leveling: UBL
Recv:  (disabled)
Recv: 
Recv: Raise Z (before homing) to 5.00
Recv: >>> do_blocking_move_to(0.00, 0.00, 5.00)
Recv: <<< do_blocking_move_to
Recv:  T:25.51 /0.00 B:23.87 /0.00 @:0 B@:0
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv: >>> do_blocking_move_to(-483.75, -498.75, 5.00)
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:  T:25.51 /0.00 B:23.87 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -483.75, [50.00])
Recv:  T:25.51 /0.00 B:23.87 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, 5.00, [50.00])
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, -10.00, 25.00)
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: Axis X home_offset = 0.00 position_shift = 0.00
Recv: > home_offset[X] = 0.00
Recv:   current_position=(0.00, 0.00, 5.00) :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -498.75, [50.00])
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, 5.00, [50.00])
Recv:  T:25.51 /0.00 B:23.87 /0.00 @:0 B@:0
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, -10.00, 25.00)
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: Axis Y home_offset = 0.00 position_shift = 0.00
Recv: > home_offset[Y] = 0.00
Recv:   current_position=(0.00, 0.00, 5.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: >>> homeaxis(Z)
Recv:   current_position=(0.00, 0.00, 5.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to(0.00, 0.00, 10.00)
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv: >>> do_blocking_move_to(0.00, 0.00, 10.00)
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -525.00, [4.00])
Recv:  T:25.51 /0.00 B:23.87 /0.00 @:0 B@:0

Roxy-3D, 2019-08-17:
I posted the request for the G29 W information in the wrong place...   I'm bringing @MarkJB 's information into this thread...

With #define UBL_DEVEL_DEBUGGING enabled, G29 W gives me the following:

    [16:01:32:696] echo:Hardcoded Default Settings Loaded?
    [16:01:32:697] ok?
    [16:01:36:762] echo:Settings Stored (614 bytes; crc 15129)?
    [16:01:36:779] ok?
    [16:01:40:781] echo:DEBUG:INFO,ERRORS,DRYRUN,COMMUNICATION,LEVELING?
    [16:01:40:782] ok?
    [16:01:57:808] Unified Bed Leveling System v1.01 inactive?
    [16:01:57:852] No Mesh Loaded.?
    [16:01:57:904] planner.z_fade_height : 0.0000?
    [16:01:57:910] # of samples: 9?
    [16:01:57:910] Mean Mesh Height: 0.000000?
    [16:01:57:910] Standard Deviation: 0.000000?
    [16:01:57:910] zprobe_zoffset: -5.6999998?
    [16:01:57:910] MESH_MIN_X (_MAX((((200) / 2) - (200) / 2) + 25, -35))=25?
    [16:01:57:954] MESH_MIN_Y (_MAX((((200) / 2) - (200) / 2) + 25, -2))=25?
    [16:01:58:006] MESH_MAX_X (_MIN((((200) / 2) + (200) / 2) - (25), 200))=175?
    [16:01:58:056] MESH_MAX_Y (_MIN((((200) / 2) + (200) / 2) - (25), 200))=175?
    [16:01:58:107] GRID_MAX_POINTS_X 3?
    [16:01:58:159] GRID_MAX_POINTS_Y 3?
    [16:01:58:209] MESH_X_DIST 75.00?
    [16:01:58:211] MESH_Y_DIST 75.00?
    [16:01:58:260] X-Axis Mesh Points at: 25.000 100.000 175.000 ?
    [16:01:58:336] Y-Axis Mesh Points at: 25.000 100.000 175.000 ?
    [16:01:58:413] ?
    [16:01:58:463] ubl_state_at_invocation :0?
    [16:01:58:464] ubl_state_recursion_chk :0?
    [16:01:58:515] Meshes go from 0x000002E8 to 0x00000F7F?
    [16:01:58:566] sizeof(ubl) : 1?
    [16:01:58:566] ?
    [16:01:58:578] z_value[][] size: 36?
    [16:01:58:578] ?
    [16:01:58:591] EEPROM free for UBL: 0x00000C97?
    [16:01:58:643] EEPROM can hold 89 meshes.?
    [16:01:58:644] ?
    [16:01:58:667] Unified Bed Leveling sanity checks passed.?
    [16:01:58:679] ok?

Roxy-3D, 2019-08-17:
I think we need your Configuration.h and Configuration_adv.h files.    I think it is possible that the 
```
#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
```
Is too small for how large of Z_PROBE_OFFSET you have, and for how big of a Z_CLEARANCE numbers you have.

Roxy-3D, 2019-08-17:
I think we are seeing the failures because of these settings:
```
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     5 // Z Clearance between multiple probes
//#define Z_AFTER_PROBING           5 // Z position after probing is done
#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
```
Let's increase (the magnitude) of Z_PROBE_LOW_POINT to -10 or -15.      That should give the machine enough room to actually activate the probe.

2kingsrock, 2019-08-17:
On my COREXY,  I turned off the UBL, and am using #define AUTO_BED_LEVELING_BILINEAR.  I have it trying a simple 3 x 3 grid.  Similar behavior to what I had in UBL.  It will probe the same 3 spots X0, X165, X330. Then thinks it moves in Y, when it stays at Y0, and repeats the X samples.

Send: G29 P1
Recv: echo:G29 P1
Recv:   current_position=(274.00, 314.00, 0.96) : >>> G29    // started at far edge of bed.  I assumed it would respect the mesh and bed areas.
Recv: Machine Type: Core
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X:45 Y:15 Z:2.00 (Right-Back & Above Nozzle)
Recv: Auto Bed Leveling: BILINEAR
Recv:  (disabled)
Recv: 
Recv:   current_position=(274.00, 314.00, 0.96) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to(274.00, 314.00, 10.00)
Recv: <<< do_blocking_move_to
Recv:  T:28.36 /0.00 B:26.88 /0.00 @:0 B@:0
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: >>> do_blocking_move_to(274.00, 314.00, 10.00)
Recv: <<< do_blocking_move_to
Recv: reset_bed_level
Recv: >>> probe_pt(45.00, 15.00, raise, 0, probe_relative)
Recv:   current_position=(274.00, 314.00, 10.00) :
Recv: >>> do_blocking_move_to(0.00, 0.00, 10.00)
Recv: <<< do_blocking_move_to
Recv:   current_position=(0.00, 0.00, 10.00) : set_probe_deployed  // At this point the system is well off the bed to the far right (340mm from 0,0 home)
Recv: deploy: 1
Recv:   current_position=(0.00, 0.00, 10.00) : >>> run_z_probe
Recv:   current_position=(0.00, 0.00, 10.00) : >>> do_probe_move

I know the bed level, auto bed level is complex and non trival.  I am going to need to spend some time reading up on this.  Going back to a manual 25pt level :(

MarkJB, 2019-08-18:
Thanks! That worked. 

Set `Z_PROBE_LOW_POINT` to -10 and head is now moving in x & y.

I'd ignored these settings as the comments talk about deploying a probe, so I wasn't expecting that setting to affect a fixed probe that is physically above the nozzle but has its trigger point is below the nozzle.

I will do some more testing with more grid points later today.

MarkJB, 2019-08-18:
I've increased the number of probe points and tweaked the mesh inset and probe deploy clearances etc and everything is still working as expected so far.

> [15:48:21:324] Bed Topography Report:␊
[15:48:21:333] ␊
[15:48:21:333]     ( 10,190)                      (190,190)␊
[15:48:21:338]         0       1       2       3       4      ␊
[15:48:21:347]  4 | -5.038  -5.130  -5.250  -5.361  -5.471 ␊
[15:48:21:387]    |␊
[15:48:21:387]  3 | -5.038  -5.075  -5.123  -5.169  -5.160 ␊
[15:48:21:387]    |␊
[15:48:21:387]  2 | -5.036  -5.025 [-5.003] -4.935  -4.818 ␊
[15:48:21:398]    |␊
[15:48:21:398]  1 | -5.051  -4.950  -4.858  -4.718  -4.490 ␊
[15:48:21:398]    |␊
[15:48:21:398]  0 | -5.069  -4.866  -4.693  -4.481  -4.252 ␊
[15:48:21:398]         0       1       2       3       4      ␊
[15:48:21:405]     ( 10, 10)                      (190, 10)␊
[15:48:21:411] ␊
[15:48:21:419] ok␊

Roxy-3D, 2019-08-18:
@MarkJB    Mark, I think you will get better results if you get your Z_PROBE_OFFSET_FROM_EXTRUDER set better.   It is claiming your entire mesh is -5 mm low.      Assuming you are homing in the center of the bed, try to get that point very close to 0.000 mm on the mesh.

MarkJB, 2019-08-19:
I'd got as far as running mesh validation and the head was crashing into the bed, so that would explain that. 

Yet cura sliced prints were slightly off the bed...

I'd kinda figured it was something to do with a z offset. So will look at that next.

Thanks for you help.

MarkJB, 2019-08-19:
By way of explanation, that z offset may have been correct at some point recently (probably with the previous IR probe) but the current inductive probe has been moved around a few times. 

When I looking at the z height after getting UBL working the nozzle was probably less than 1mm above the bed.

MarkJB, 2019-08-30:
Just spent some time looking at the Z_PROBE_OFFSET_FROM_EXTRUDER which is set in firmware to -0.7. When the probe triggers, the nozzle is 0.7m above the bed, but when I run the G29 P1 I get those ~5mm values. 

Is there another offset somewhere that is adding that ~4.3mm ?

I set the z probe offset to +4.3 with 'M851 Z4.3' and ran G29 P1 and my readings were all closer to 0 than the above but homing Z puts the nozzle way above the bed now.

MarkJB, 2019-08-30:
Seems like there is a 5mm offset in there somewhere but I can't see where - guess I should open another bug?

Roxy-3D, 2019-08-30:
That 5mm offset should be a direct result of the Z_PROBE_OFFSET_FROM_EXTRUDER being wrong.   
When you home the machine....    Do you have to lower the nozzle 5mm to get it to just touch the bed?

Any way...   shifting the Z_PROBE_OFFSET_FROM_EXTRUDER value by 5 mm. (and doing a M502 & M500) should get you much closer to being correct.

Like I say...   The test is home the machine...   and very slowly lower the nozzle until it just touches the bed.   The LCD Panel (or M114 if you don't have an LCD Panel) should report a value very close to 0.00 mm if you have the offset declared correctly.

MarkJB, 2019-09-01:
With Z_PROBE_OFFSET_FROM_EXTRUDER 6 the hotend is 1mm above the bed after homing and the LCD shows Z=1. I can drop the head down to zero and it just grabs a sheet of paper so that seems good.

But running G29 P1 T shows me ~ -5 values again...

When I try to print the mesh validation shape it crashes the head into the bed.

What I can't wrap my head around is the point at which the probe triggers when homing puts the nozzle exactly 1mm above the bed so can't understand why the Z_PROBE_OFFSET_FROM_EXTRUDER isn't -1

MarkJB, 2019-09-02:
I had my Z_MIN_POS set to -5 (to give me enough leeway so I thought, to adjust the probe offset while I was still setting up the printer). Set it to 0 and the mesh reports nearer 0 values.

After homing the LCD was reporting the z height as 6 which reflected the -6 of Z_PROBE_OFFSET_FROM_EXTRUDER. Set that to -1 and I'm back to +1 on the LCD after homing.

I wasn't expecting Z_MIN_POS to affect the probe offset.

Mesh validation printed without ramming the head into the bed.

closed 2019-09-07 by MarkJB