MarlinFirmware/Marlin #14980: MKS Sbase CR10 LCD

lightface79, 2019-08-18:
Trying to get my CR10 unerversial full graphics lcd to work with my MKS Sbase 1.3 and latest marlin 2 build

### Steps to Reproduce

1. Board definde lpc1768
2. Uncomment Config.h  #define CR10_STOCKDISPLAY
3. Build fw without errors

**Expected behavior:** [What you expect to happen]

Screen should work showing text/graphics
Enconder should work / Button encoder

**Actual behavior:** [What actually happens]
LCD backlight on
Beeps during boot / quiet after boot

#### Additional Information

The LCD have only 1 cable/contact  (Ext1(   Connected 180Degrees wrong on sbase on EXT1
Tried rotating the ext1 with no luck just dark lcd without Backlight

[image]
[image]
[image]

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3512913/Marlin.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

lightface79, 2019-08-18:
Im looking in the pins_mks_sbase.h   
and just compare to the pins_mks_sgen  
i see a big difference...

in the lcd part of the mks_sbase im missin somthin like this that is located in the mks_sgen

/**
 *                _____                                            _____
 * (BEEPER) 1.31 | · · | 1.30 (BTN_ENC)          (MISO)       0.8 | · · | 0.7  (SD_SCK)
 * (LCD_EN) 0.18 | · · | 0.16 (LCD_RS)           (BTN_EN1)   3.25 | · · | 0.28 (SD_CS2)
 * (LCD_D4) 0.15 | · · | 0.17 (LCD_D5)           (BTN_EN2)   3.26 | · · | 1.20 (SD_MOSI)
 * (LCD_D6)  1.0 | · · | 1.22 (LCD_D7)           (SD_DETECT) 0.27 | · · | RST
 *           GND | · · | 5V                                   GND | · · | NC
 *                -----                                            -----
 *                EXP1                                             EXP2
 */
#if HAS_SPI_LCD
  #define BEEPER_PIN       P1_31
  #define BTN_ENC          P1_30

  #if ENABLED(CR10_STOCKDISPLAY)
    #define LCD_PINS_RS    P1_00

    #define BTN_EN1        P0_18
    #define BTN_EN2        P0_15

    #define LCD_PINS_ENABLE P1_22
    #define LCD_PINS_D4    P0_17

  #else
    #define LCD_PINS_RS    P0_16

    #define BTN_EN1        P3_25
    #define BTN_EN2        P3_26

    #define LCD_PINS_ENABLE P0_18
    #define LCD_PINS_D4    P0_15

    #define LCD_SDSS       P0_28
    #define SD_DETECT_PIN  P0_27

    #if ENABLED(FYSETC_MINI_12864)
      #define DOGLCD_CS    P0_18
      #define DOGLCD_A0    P0_16
      #define DOGLCD_SCK   P0_7
      #define DOGLCD_MOSI  P1_20
      #define FORCE_SOFT_SPI

      #define LCD_BACKLIGHT_PIN -1

the hole pin asignment is missing for the lcd in the mks_sbase

[pins_MKS_SBASE.txt](https://github.com/MarlinFirmware/Marlin/files/3513235/pins_MKS_SBASE.txt)

thisiskeithb, 2019-08-20:
Looking at the [MKS Sbase pinout](https://github.com/makerbase-mks/MKS-SBASE/blob/160580d0e9ddded5058e3af65baa55304e59e3b9/Hardware/MKS%20SBASE%20V1.3_002/MKS%20SBASE%20V1.3_002%20PIN.pdf), I don't think you'll be able to use a single cable LCD like the `CR10_STOCKDISPLAY` without physical modification.

`CR10_STOCKDISPLAY` pins:
[image]

lightface79, 2019-08-20:
[CR10 pinouts mks sbase.txt](https://github.com/MarlinFirmware/Marlin/files/3519186/CR10.pinouts.mks.sbase.txt)

lightface79, 2019-08-20:
Started to pin out everything between the EXP1 And 2 But i still cant get a pixel to apear on the screen =(

lightface79, 2019-08-20:
CR10 full graphics Pinout Single cable   EXP1
NO SD card reader

	P1.31	Beeper + 2 O		O   1 BTN_ENC		P1.30      (Red strand/cable)
 
	P3.26	BTN ENC1 4 O		O   3 (NC) S2 BUTTOn	NC			

	P3.25	BTN Enc2 6 O ]		O   5 LCD E		P0.18
		
	P0.16	LCD_RS   8 O		O   7 LCD R/W           ?????  what is this??  

		GND	10 O		O   9 Vin 5v+

PINOUT MKS SBASE 1.3     EXP1/EXP2

EXP1

1 P1.30
2 P1.31
3 P0.16
4 P0.18
5 NC     (notch)
6 P0.15
7 NC
8 NC
9 5v+
10 GND

EXP2

1 P0.7
2 P0.8
3 P0.28
4 P3.25
5 P0.9   (NOTCH)
6 P3.26
7 RESET
8 P0.27
9 NC
10 GND

Pin Index To function

EXP1

1 P1.30 	VBUS/AD0[4]  
2 P1.31		SCK1/AD0[5]
3 P0.16		RXD1/SSEL0/SSEL
4 P0.18		DCD1/MOSI0/MOSI
5 NC
6 P0.15		TXD1/SCK0/SCK
7 NC
8 NC
9 5v+
10 GND

EXP2

1 P0.7		I2STX_CLK/SCK1/MAT2[1]
2 P0.8		I2STX_WS/MISO1/MAT2[2]
3 P0.28		SCL0/USB_SCL
4 P3.25		AD0[2]/I2SRX_SDA/TXD3
5 P0.9  	I2STX_SDA/MOSI1/MAT2[3]
6 P3.26		AD0[3]/AOUT/RXD3
7 RESET
8 P0.27		SDA0/USB_SDA
9 NC
10 GND

timh0r, 2019-08-25:
i've been there. Tried to get my CR10 display from my ender 3 to work on 1 ribbon cable with my SGen. Ended with an home made adapter, lcd would display stuff but encoder would not work.
Went with classic cabling and used exp1 + 2 instead of 3.
Take care of the pinout that is sometimes reversed on  MKS

boelle, 2019-09-24:
@lightface79 do you still have a problem?

shitcreek, 2019-10-03:
Hardware issue.

closed 2019-10-03 by shitcreek