MarlinFirmware/Marlin #14982: [BUG] Delta auto calibration fails

McVillano, 2019-08-18:
### Description

When I try to Configuration->Delta Calibration->Auto Calibration it makes the first iteration without a problem but in the second iteration (sometimes the third) one of the probe points goes down less than normal, doesn't touch and fails. Video Attached: https://drive.google.com/file/d/1-cl7ujITy2jqdxIsU8bRm4PV8ut8zMFv/view?usp=sharing

### Steps to Reproduce

Compile with the settings attached and try to Auto Caibrate a delta printer 

**Expected behavior:** Auto Adjuts

**Actual behavior:** Probbing Failed

#### Additional Information

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3513017/Marlin.zip)
 It's a SKR 1.3 LPC1768

Bergerac56, 2019-08-18:
Did you modify some hardware elements affecting the DELTA_HEIGHT ? (Like the position of the endstops etc...)
To be sure that it is not related to the DELTA_HEIGHT, try to compile with DELTA_HEIGHT a bit higher (301.0 for example) and see if it helps. 

Do not forget that: "// NOTE NB all values for DELTA_* values MUST be floating point, so always have a decimal point in them". I see some DELTA_* values without decimal point...

McVillano, 2019-08-18:
It's the first mount of the printer so i have changed all jejejeje.

I've started compiling with 320 delta height and i went down to 310 and then 300 and It has the same result. Also after the first iteration it gets adjusted to arround 275. This is why i think it's not delta height problem. But i'm going to put floating point and recompile, maybe it works

McVillano, 2019-08-18:
> Did you modify some hardware elements affecting the DELTA_HEIGHT ? (Like the position of the endstops etc...)
> To be sure that it is not related to the DELTA_HEIGHT, try to compile with DELTA_HEIGHT [...]

I've tried with DELTA HEIGHT 330.0 and no luck, for the first time i managed to get to the 4 iteration but no luck, after that i've re-done it and it failed at 2º as always. 

A strange thing to me is that when i start the calibration the screen says "($MachineName) Ready" and when it finishes and goes for the next it says "Iteration 01"

Maybe this firts "measurement" is messing up everything?

Bergerac56, 2019-08-19:
This could be. It seems  like one parameter goes out of range after the first iteration. Perhaps because too close to a limit. If you connect the printer to Pronterface for ex, you will get a bit more information on what the printer is doing. It could help. 
I do not know if this is useful as I do not have a anycubic, but here is my "config" file

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3514497/Configuration.zip)

McVillano, 2019-08-19:
It was a stepper problem it was doing wrong movements because off to much acceleration for the driver and the driver comming hot and loosing steeps with 2k acell and blowing the drivers all the time i managed to finish. I have ordered a blower jejejeje
THANKS!

closed 2019-08-19 by McVillano