MarlinFirmware/Marlin #14984: Marlin 2.0 homing problem (MKS SBASE)

renzaaa, 2019-08-18:
Homing normally works just in first time. In second time homing don't work normally.
For example:
G28 X0 - homing normally to 0 postition, then repeating G28 X0 comand X axis moving to + position.

boelle, 2019-08-19:
please attach your config as the issue template said assuming you read that one

renzaaa, 2019-08-19:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3517101/Config.zip)

TrollNick, 2019-08-27:
I have a similar problem.
Do you use TMC Stepper, and maybe sensorless homing?
For me the problem occurs after i finished a print. When i start a 2nd print homing works just fine, when the printer then wants to lay down the first line he goes to what seems the opposide coordinates (Up instead of down, left instead of right and back instead of front).
When i reset the printer (Reset button or cycle power), the printer prints normally as expected. So after every print i have to reset the board to have a functioning printer

tpruvot, 2019-08-27:
Seen the same here, seems recent. After a first SD print, the printer seems "locked" and requires a reboot

renzaaa, 2019-08-27:
No, i'm using drv8825.
After every print i have to reset the board too.
Still don't know where the problem is.

TrollNick, 2019-08-27:
I have DRV8825 for my Z and E too.
Do you have other Drivers lying around to rule out the DRV Drivers?

Here #14949 is another Issue i opend about the problem 2 weeks ago. Made some tests today and posted the results there

> Here is what i tested today:
>
>Started machine
>Finished a print
>Sent G28 via Octoprint
>Moved axis via display. Everything was exactly as expected. Axis moved in the right direction and
>position on the display was aligned with the real position of the nozzle.
>Started a new print.
>Same faulty behavior occured (Ran into machine limits and extruded like crazy after home went >through without a problem)
>Made a 2nd run:
>
>Reset Printer
>Finish a print
>Start the same print again
>Homing worked just fine, Position right after homing was correct
>Then the printer moved to the Machine limits (X_MAX_POS and Y_MAX_POS) and showed these on the Display (X230 Y220 X13.33) So the position of the nozzle aligns with what is shown in the display.
>But ofcourse the GCode does not tell the machine to go to X230 Y220. It seems like the machine just goes "You know what i dont care i go to the Limits and just extrude."

sjasonsmith, 2019-08-31:
I attempted to reproduce this on my machine. Repeated homing works fine with my sBase (and its on-board DRV8825 drivers). I'm building against 87b16ed, which is only a few hours old.

I can't use the config @renzaaa because I don't have a CoreXY machine, and I invert my axis/endstop values differently. I don't see anything different in the way the actual stepper drivers are configured.

boelle, 2019-09-24:
@renzaaa is this still a problem?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

renzaaa, 2019-10-14:
The issue is not solved. 
I tried the latest bugfix, but problem not gone.
Delta 3D printer is with same problem (with mks sbase board too).

boelle, 2019-10-14:
any idea what the error could be? only @tpruvot seems to be able to re-produce the error

tpruvot, 2019-10-14:
euh no, was fixed the same day or so, for mine

renzaaa, 2019-10-14:
See the video on this link:
https://drive.google.com/drive/folders/1TUZgPJR3r-J2bbhrVIqr3aU9XtIEPfUP?usp=sharing

boelle, 2019-10-14:
what happens when you send M119 and no endstop is touching?

renzaaa, 2019-10-14:
M119
READ: x_min: open
READ: y_min: open
READ: z_min: open

boelle, 2019-10-14:
and when you touch the endstops i guess they change to close, ie when you touch X with finger it changes to close

i had the same problem once but it was because they was connected wrong

renzaaa, 2019-10-14:
After homing x axis:
READ: x_min: TRIGGERED
READ: y_min: open
READ: z_min: open
Then repeat homing x axis:
The x axis move to + position and still x_min: TRIGGERED.
If endstop is not touched after a first time of homing, endstop still to be triggered or to be open as well?

renzaaa, 2019-10-14:
Sorry for my English language.

boelle, 2019-10-14:
if endstop is not touched after homing it should be open

try send M84
then carefull with hand move nozzle to middle of bed and half way of Z travel
send M119 and check that all endstop are open

Send G28 X so that only X is homed

Send M119 again

is X endstop triggered?

if yes try send G0 X10 and send M119 and see if still triggered

sjasonsmith, 2019-10-14:
> I tried the latest bugfix, but problem not gone.

Can you post the updated configs you tested with?

renzaaa, 2019-10-14:
I will try

renzaaa, 2019-10-14:
[Config (2).zip](https://github.com/MarlinFirmware/Marlin/files/3725796/Config.2.zip)

sjasonsmith, 2019-10-14:
I was going to try it out, but just realized that I already tried to reproduce this issue the first time around. I don't have a machine similar enough to get any closer to your setup than what I've already done.

renzaaa, 2019-10-14:
to sjasonsmith
My delta 3D printer with same problem

tpruvot, 2019-10-14:
would not be the first time this issue reappears...

renzaaa, 2019-10-14:
to boelle
I need to restart my 3D printer, because can't repeat the X axis homing.
After restart:
M119
READ: x_min: open
READ: y_min: open
READ: z_min: open
G28 X/M119
READ: x_min: TRIGGERED
READ: y_min: open
READ: z_min: open

boelle, 2019-10-14:
so you cant send G0 X10 after M119 ?

renzaaa, 2019-10-14:
G0 X10 - move 10mm 
M119 - still triggered

boelle, 2019-10-14:
something is wrong with your endstop, that is why i fails

boelle, 2019-10-14:
can you attach a picture of your board with endstops connected?

ie take picture with your phone and insert here

boelle, 2019-10-14:
try move nozzle to the middle again
send M119 and check they are all open

put your finger on the endstop, do not home 

send M119 again and check it changes to triggered

remove your finger and send M119 again and check it says open

in that way we know its not your endstop that is bad

renzaaa, 2019-10-14:
[image]

boelle, 2019-10-14:
what did the finger test say?

renzaaa, 2019-10-14:
Finger test:
READ: x_min: TRIGGERED

boelle, 2019-10-14:
and if you remove finger and send M119 again?

renzaaa, 2019-10-14:
Still triggered

boelle, 2019-10-14:
your endstop is bad

what if you do the same finger test with Y endstop?

ie put finger on Y endstop send M119 
remove finger from y endstop and send M119 again?

renzaaa, 2019-10-14:
All axis endstops with same problem on corexy 3D printer.
On delta 3d printer is too.

boelle, 2019-10-14:
we need to keep focus one machine at a time

did you do the finger test on Y axis ?

renzaaa, 2019-10-14:
Yes, but like x axis is triggered

boelle, 2019-10-14:
hmm.... what endstop voltage is the board set to?

boelle, 2019-10-14:
[image]

renzaaa, 2019-10-14:
I was try from 5v change to 3,3v, but nothing to changed

renzaaa, 2019-10-14:
Now is set to 5v

Patag, 2019-10-14:
I've this kind of issue also on a SKR1.3 using sensorless homing (TMC5160). 
It looks like a wrong X position (maybe 0) is memorized somewhere and is used in the next Auto Home command.

boelle, 2019-10-14:
i will try and look at the config files to see if something comes to my mind

renzaaa, 2019-10-14:
But my EEPROM is disabled.

renzaaa, 2019-10-14:
Ok, thanks boelle

boelle, 2019-10-14:
i cant see anything wrong with the config, and it would be very strange if your endstops are all bad

what type of endstop is it?

boelle, 2019-10-14:
can you post a pitcture of the endstop?

renzaaa, 2019-10-14:
Z axis endstop works normally. 
I tested just now.
Problem is with XY.
Ok, one minute..

boelle, 2019-10-14:
what if you connect Z endstop to where X endstop is connected, does X work normal when doing the finger test?

renzaaa, 2019-10-14:
[image]
[image]
I will try

renzaaa, 2019-10-14:
Problem with endstops or endstop brackets. Maybe problem with contacts, 
I plugged another endstop and works normally.

renzaaa, 2019-10-14:
Very strange..

boelle, 2019-10-14:
maybe endstop is bad? or wire from endstop to board is bad? or is wire the same and you just changed endstop?

renzaaa, 2019-10-14:
Yes 
I removed endstop from bracket and endstop is bad :/ I don't know how broken my endstops.
Problem is solved, thanks for helping!

boelle, 2019-10-14:
maybe its just bad luck :-)
or maybe its bad quality from china

renzaaa, 2019-10-14:
Maybe :) 
Most importantly that we found a problem.
Thanks.

renzaaa, 2019-10-14:
No no, endstops is not bad, just dust came into contacts :D