MarlinFirmware/Marlin #14989: Senroless homing with TMC 2130 stop immediately

RobertPrudek, 2019-08-18:
Hello, 
i have problem with sensorless homing with TMC2130. When i try home X or Y it immediately triggers and stop, or it end up with Error: Printer halted. Kill () called. Motor make some noise (like he is trying to move but don't have enough power). If it triggers immediately then i can move motor and it moves (quietly becouse i enabled STEALTHCHOP), so it have enough power.

I'm using marlin 1.1.9, on ramps 1.6 and SPI is wired to HW SPI, but i think that i change it to SW SPI, becouse of LCD (while trying find answer on my problem i found out that it can be wired to SW SPI).

These parameters i changed to 300 or higher (and i think that is high enough)
> ```c++
> #define SENSORLESS_HOMING // TMC2130 only
>
> #if ENABLED(SENSORLESS_HOMING)
>    #define X_HOMING_SENSITIVITY  10
>    #define Y_HOMING_SENSITIVITY  10
>    //#define Z_HOMING_SENSITIVITY  8
>  #endif
> ```

While searching i found instructions on TMC2130 (which is on marlin web site) and found out that i dind't done one think and that is make 2 bridges on drivers, now i can't try if it could be that problem, becouse i'm not home. I'll be home at friday.

If i send m122 
>20:06:34.312 : X	Y
>20:06:34.312 : Enabled		false	false
>20:06:34.312 : Set current	800	800
>20:06:34.322 : RMS current	795	795
>20:06:34.322 : MAX current	1121	1121
>20:06:34.322 : Run current	25/31	25/31
>20:06:34.322 : Hold current	12/31	12/31
>20:06:34.322 : CS actual		12/31	12/31
>20:06:34.322 : PWM scale		80	72
>20:06:34.332 : vsense		1=.18	1=.18
>20:06:34.332 : stealthChop	true	true
>20:06:34.332 : msteps		16	16
>20:06:34.332 : tstep		1048575	1048575
>20:06:34.332 : pwm
>20:06:34.332 : threshold		79	79
>20:06:34.342 : [mm/s]		100.08	100.08
>20:06:34.342 : OT prewarn	false	false
>20:06:34.342 : OT prewarn has
>20:06:34.342 : been triggered	false	false
>20:06:34.342 : off time		5	5
>20:06:34.342 : blank time	24	24
>20:06:34.342 : hysteresis
>20:06:34.352 : -end		2	2
>20:06:34.352 : -start		3	3
>20:06:34.352 : Stallguard thrs	10	10
>20:06:34.352 : DRVSTATUS	X	Y
>20:06:34.352 : stallguard
>20:06:34.352 : sg_result		91	89
>20:06:34.352 : fsactive
>20:06:34.362 : stst		X	X
>20:06:34.362 : olb
>20:06:34.362 : ola			X
>20:06:34.362 : s2gb
>20:06:34.362 : s2ga
>20:06:34.362 : otpw
>20:06:34.362 : ot
>20:06:34.362 : Driver registers:	X = 0x80:0C:00:5B
>20:06:34.362 : Y = 0xA0:0C:00:59

If you have any question feel free to ask.

boelle, 2019-08-19:
try with 2.0.x
and as the issue template said attach your config

RobertPrudek, 2019-08-24:
So guys problem was in that 2 little bridges, after i made them printer started working properly. So the problem was between the chair and keyboard. I tryed to configure it with help of some youtube video and he said to destroy that one bridge, but he forget to tell there should be that small 2 bridges as you show in your web site [http://marlinfw.org/docs/hardware/tmc_drivers.html](url). But thank you for your help.

closed 2019-08-24 by RobertPrudek