MarlinFirmware/Marlin #14993: Max temp only 260

deezdrama, 2019-08-19:
### Description

Creality 1.1.5 board with newest marlin 1.1.x bugfix.
My max nozzle temp can now only be increased to 260c. Even If gcode is sliced above 260c the ender will max at 260c. I have all metal hot end and need more temp for polycarbonate.
Maybe this is a simple fix but im still fairly new to editing the firmware files etc.... and after hours of searching I cant find any info or examples of this same issue anywhere.
1.1.8 worked fine but homing speeds were too fast so I flashed 1.1.x bugfix and now this.

My apologies if this is not a bug but been searching all day for an answer.
Thanks

teemuatlut, 2019-08-19:
The issue template asks you to include your config files, but you can start by looking
at your `HEATER_0_MAXTEMP` in Configuration.h:353.

robbycandra, 2019-08-19:
It is not a bug... It is a safety made by marlin.

AnHardt, 2019-08-19:
Set maximum temperature for tat heater to the max(*15°C above the temperature you want to print with* , *the maximum temperature your hotend can stand for a long time*).
15°C is a safety margin against maxtemp-errors because of temperature overshoot during the end of initial heating. It's needed, but the exact value is discussable.

boelle, 2019-08-19:
also use bugfix 2.0.x

1.1.x is not updated anymore

boelle, 2019-08-19:
and as @teemuatlut said attach your config, the issue template if you read it does ask to include config

deezdrama, 2019-08-19:
Do you just want me to paste the whole config file contents to my first post or just the maxtemp settings section?

I downloaded 1.1.x bugfix 2-3 weeks ago, used the included ender 3  example config files and used them to overwrite the main marlin files. Then used arduino software to add sanguino board , selected 1284p processor, added u8glib 1.17 library, then compiled.

Did I do something wrong?
The maxtemp section of the example configH file states...
"// When temperature exceeds max temp, your heater will be switched off.
// This feature exists to protect your hotend from overheating accidentally, but *NOT* from thermistor short/failure!
// You should use MINTEMP for thermistor short/failure protection.
#define HEATER_0_MAXTEMP 275
#define HEATER_1_MAXTEMP 275
#define HEATER_2_MAXTEMP 275
#define HEATER_3_MAXTEMP 275
#define HEATER_4_MAXTEMP 275
#define BED_MAXTEMP 125"

So because my max is set at 275 but my machine will only go to 260, so it is defaulted to a 15 less safety margin? Im pretty sure my other board and 2 older firmware versions went to 280 or 300.
Why is it set this low? Ive never read of anyone else having this issue, my buddies goes up to 300 , he said he flashed same firmware and config.

boelle, 2019-08-19:
no attach the 2 files as zip

deezdrama, 2019-08-19:
[Ender-3.zip](https://github.com/MarlinFirmware/Marlin/files/3517947/Ender-3.zip)

AnHardt, 2019-08-19:
Why is this issue still open and discussed?

With the provided MAXTEMP settings the described behavior is wanted one.

Please close.

AnHardt, 2019-08-19:
@boelle 
Asking for more data when the solution is already there is mad.

deezdrama, 2019-08-19:
So if I want to run 275 then I just change to.....

#define HEATER_0_MAXTEMP 290
#define HEATER_1_MAXTEMP 275
#define HEATER_2_MAXTEMP 275
#define HEATER_3_MAXTEMP 275
#define HEATER_4_MAXTEMP 275
#define BED_MAXTEMP 125"

?

deezdrama, 2019-08-19:
And im sorry Ive been asking for info on this across 4 different forums and noone knows anything. I just remembered max temp going higher on an older version

So I just need to change #define HEATER_0_MAXTEMP 290 ?

Thanks

AnHardt, 2019-08-19:
If your hotend can stand that for a short time - yes.

closed 2019-08-19 by deezdrama