MarlinFirmware/Marlin #14995: [BUG] (short description)

RobMink, 2019-08-19:
### Description

SCARA will not compile in Bugfix 2.0

### Steps to Reproduce

1. Stock config for SCARA but change to serial -1 and boards to radds
2. Set board to DUE in Ardunio IDE 1.8.9

**Expected behavior:** Compilation

**Actual behavior:** 
In file included from sketch\src\gcode\control\../../module/planner.h:35:0,

                 from sketch\src\gcode\control\../../module/stepper.h:227,

                 from sketch\src\gcode\control\M17_M18_M84.cpp:26:

sketch\src\gcode\control\../../module/motion.h:259:24: error: redeclaration 'L1' differs in 'constexpr'

     extern const float L1, L2;

                        ^

In file included from sketch\src\gcode\control\../../module/motion.h:34:0,

                 from sketch\src\gcode\control\../../module/planner.h:35,

                 from sketch\src\gcode\control\../../module/stepper.h:227,

                 from sketch\src\gcode\control\M17_M18_M84.cpp:26:

sketch\src\gcode\control\../../module/scara.h:33:17: error: from previous declaration 'L1'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                 ^

In file included from sketch\src\gcode\control\../../module/planner.h:35:0,

                 from sketch\src\gcode\control\../../module/stepper.h:227,

                 from sketch\src\gcode\control\M17_M18_M84.cpp:26:

sketch\src\gcode\control\../../module/motion.h:259:28: error: redeclaration 'L2' differs in 'constexpr'

     extern const float L1, L2;

                            ^

In file included from sketch\src\gcode\control\../../module/motion.h:34:0,

                 from sketch\src\gcode\control\../../module/planner.h:35,

                 from sketch\src\gcode\control\../../module/stepper.h:227,

                 from sketch\src\gcode\control\M17_M18_M84.cpp:26:

sketch\src\gcode\control\../../module/scara.h:33:39: error: from previous declaration 'L2'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                                       ^

In file included from sketch\src\gcode\control\M211.cpp:28:0:

sketch\src\gcode\control\../../module/motion.h:259:24: error: redeclaration 'L1' differs in 'constexpr'

     extern const float L1, L2;

                        ^

In file included from sketch\src\gcode\control\../../module/motion.h:34:0,

                 from sketch\src\gcode\control\M211.cpp:28:

sketch\src\gcode\control\../../module/scara.h:33:17: error: from previous declaration 'L1'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                 ^

In file included from sketch\src\gcode\control\M211.cpp:28:0:

sketch\src\gcode\control\../../module/motion.h:259:28: error: redeclaration 'L2' differs in 'constexpr'

     extern const float L1, L2;

                            ^

In file included from sketch\src\gcode\control\../../module/motion.h:34:0,

                 from sketch\src\gcode\control\M211.cpp:28:

sketch\src\gcode\control\../../module/scara.h:33:39: error: from previous declaration 'L2'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                                       ^

In file included from sketch\src\Marlin.cpp:35:0:

sketch\src\module/motion.h:259:24: error: redeclaration 'L1' differs in 'constexpr'

     extern const float L1, L2;

                        ^

In file included from sketch\src\module/motion.h:34:0,

                 from sketch\src\Marlin.cpp:35:

sketch\src\module/scara.h:33:17: error: from previous declaration 'L1'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                 ^

In file included from sketch\src\Marlin.cpp:35:0:

sketch\src\module/motion.h:259:28: error: redeclaration 'L2' differs in 'constexpr'

     extern const float L1, L2;

                            ^

In file included from sketch\src\module/motion.h:34:0,

                 from sketch\src\Marlin.cpp:35:

sketch\src\module/scara.h:33:39: error: from previous declaration 'L2'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                                       ^

exit status 1
In file included from sketch\src\gcode\gcode.cpp:33:0:

sketch\src\gcode\../module/motion.h:259:24: error: redeclaration 'L1' differs in 'constexpr'

     extern const float L1, L2;

                        ^

In file included from sketch\src\gcode\../module/motion.h:34:0,

                 from sketch\src\gcode\gcode.cpp:33:

sketch\src\gcode\../module/scara.h:33:17: error: from previous declaration 'L1'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                 ^

In file included from sketch\src\gcode\gcode.cpp:33:0:

sketch\src\gcode\../module/motion.h:259:28: error: redeclaration 'L2' differs in 'constexpr'

     extern const float L1, L2;

                            ^

In file included from sketch\src\gcode\../module/motion.h:34:0,

                 from sketch\src\gcode\gcode.cpp:33:

sketch\src\gcode\../module/scara.h:33:39: error: from previous declaration 'L2'

 float constexpr L1 = SCARA_LINKAGE_1, L2 = SCARA_LINKAGE_2,

                                       ^

Error compiling for board Arduino Due (Native USB Port).

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RobMink, 2019-08-19:
[config for scara and DUE.zip](https://github.com/MarlinFirmware/Marlin/files/3514179/config.for.scara.and.DUE.zip)

RobMink, 2019-08-23:
I just found https://github.com/MarlinFirmware/Marlin/issues/10089 .  Is this true for Marlin 2.0?

sjasonsmith, 2019-09-20:
I am able to build the current bugfix-2.0.x fro DUE using the provided configs just fine. @RobMink, can you close this if it is no longer an issue?

I built it in Visual Studio Code with Platform IO, not the Arduino IDE. For the errors you reported I do not think that should make a difference.

sjasonsmith, 2019-09-20:
So it turns out that it _does_ fail in the Arduino IDE, and not PlatformIO. The GCC versions used must enforce constexpr rules slightly differently. It's an easy fix, I'll post it soon.

It won't resolve any issues with it actually _working_, but at least you will be able to compile :)

shitcreek, 2019-10-03:
@RobMink is this still an issue?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle