MarlinFirmware/Marlin #14997: SKR V1.3 + Watterott TMC 5160 V1.4

tonn0297, 2019-08-19:
Hello.
i had had some problems with the tmc 5160 v1.4 from watterott.
they have deleted the clk pin on the stepsticks. and put a DCO pin there instead. the DCO is an output if the stepper is in SPI mode, wich this is. BUT on the skr v1.3 the pin that the DCO pin is connected to is also the MOTO_MISO, aka the spi read.
i've bend the DCO pin on the stepstics out and it reads the spi again.

boelle, 2019-08-19:
just confused here, is this a marlin bug?

tonn0297, 2019-08-19:
No it's a hardware issue i just think that it shuld be noted bc a lot of People are running watterott in eu

boelle, 2019-08-19:
you should post it on discord instead as this list is for bugs only

https://github.com/MarlinFirmware/Marlin/issues/14694

closed 2019-08-19 by boelle