MarlinFirmware/Marlin #14998: [BUG] Conditionals_LCD.h not expanding #define pins in pins_BOARD.h scope

soimy, 2019-08-19:
### Description

When I tried to compile the new `bugfix-2.0.x` branch with sample Ultimaker configuration, I face the compiling error:
```
HAL_AVR/fastio_AVR.h:76:41: error: 'DIOLCD_PINS_D6_RPORT' was not declared in this scope
```
After some digging, I found the error is at https://github.com/MarlinFirmware/Marlin/blob/ed86550061ab94f10dd1d310c994bd74db582740/Marlin/ultralcd_impl_DOGM.h#L364
expanding `LCD_RESET_PIN` all the way to `DIOLCD_PINS_D6_DDR` which is set by:
https://github.com/MarlinFirmware/Marlin/blob/85c82ab28e13cb25043ae166062785bff39434b9/Marlin/Conditionals_LCD.h#L148

I think there must be some potential scope error between `Conditinals_LCD.h` and `pins_BOARD.h` but that is beyond my skill to solve.

### Steps to Reproduce

1. Use configuration files inside `config\examples\AliExpress\UM2pExt\` ;
2. Compile

soimy, 2019-08-24:
This bug is annoying! I can bypass it by defining LCD_RESET_PIN in ‘pin _BOARD.h’ but it’s not an elegant approach.

thinkyhead, 2019-08-29:
The issue is now patched.

I don't have one of these boards on hand. Do you know what the `LCD_PINS_D6` pin (pin 5) is labeled on that board or in its schematic, or which header contains pin 5?

I am asking because in the future we may want to have pins like `AUX01...AUX10` in the pins files, and then define LCD (or other) pins based on those `AUXnn` pins. That way we can, for example, assign either a reset or an LCD function to a pin without assuming it is always or primarily an LCD function pin.

closed 2019-08-29 by thinkyhead

soimy, 2019-08-31:
The answer is a bit complicated. Most RAMPS compatible board on my hand have EXP1/EXP2 connect to the LCD/SD card extension board. `LCD_PINS_RS`\`LCD_PINS_D6` is the reset pin located at EXP1 Pin7.
Below is from pin def of MKS_SGEN_L comment.
```
/**
 *              _____                                             _____
 *          NC | · · | GND                                    5V | · · | GND
 *       RESET | · · | 1.31(SD_DETECT)             (LCD_D7) 1.23 | · · | 1.22 (LCD_D6)
 *  (MOSI)0.18 | · · | 3.25(BTN_EN2)               (LCD_D5) 1.21 | · · | 1.20 (LCD_D4)
 * (SD_SS)0.16 | · · | 3.26(BTN_EN1)               (LCD_RS) 1.19 | · · | 1.18 (LCD_EN)
 *   (SCK)0.15 | · · | 0.17(MISO)                 (BTN_ENC) 0.28 | · · | 1.30 (BEEPER)
 *              -----                                             -----
 *              EXP2                                              EXP1
 */
```

Ultimaker mainboard is a bit different in EXP1 layout because it uses I2C signal to connect the UltiController which is SSD1309 i2c 12864 OLED screen. But according to the schematic, EXP1 Pin7 is also `LCD_PINS_D6` and stand for `LCD_RESET_PIN`

commit 8196ced2 (2019-08-29): Leave LCD_PINS_Dn defined for other uses

Fix #14998