MarlinFirmware/Marlin #15000: sd card

devindasrinath, 2019-08-19:
hello, i made my own 3d printer and it used marlin firmware.also i made controller panel using lcd ,encoder switch and sd card reader. i made it using reprap smart discount controller diagrams and its works great with marlin firmware,but when i insert the 2gb fat32 formatted sd card to slot, it didn't detect it,after when i connected it  with computer and open it with pronterface, i could pint things from sd card by selecting sd option,so what i do now ,i want print  things from 3d printer without connect with computer,

devindasrinath, 2019-08-19:
hello, i made my own 3d printer and it used marlin firmware.also i made controller panel using lcd ,encoder switch and sd card reader. i made it using reprap smart discount controller diagrams and its works great with marlin firmware,but when i insert the 2gb fat32 formatted sd card to slot, it didn't detect it,after when i connected it with computer and open it with pronterface, i could pint things from sd card by selecting sd option,so what i do now ,i want print things from 3d printer without connect with computer, also sorry for my bad english.

boelle, 2019-08-19:
-RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
-[Tom's 3D Forums](https://forum.toms3d.org/)
-Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
-Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
-[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
-Marlin Discord server. Join link: https://discord.gg/n5NJ59y