MarlinFirmware/Marlin #15005: [FR] Auto-retraction via Linear Advance

richfelker, 2019-08-20:
### Description

A while back in https://github.com/Ultimaker/CuraEngine/issues/1055 for Cura, I requested a feature to perform retraction concurrently with travel. It was noted at the time that this would be difficult without being able to predict the printer firmware's incorporation of acceleration and jerk to motion.

Fast-forward to now: after playing around with Linear Advance in Marlin 2.0 quite a bit, including with very high K values for flex filaments with a bowden, it struck me that Marlin is already doing something very close to the sort of "retraction concurrent with travel" that I asked for in Cura - not for semantic retractions, but rather for negative adjustment of the extruder to relieve pressure when decelerating - and that the same idea could be extended to fully automatic retraction, probably largely reusing the existing math from Linear Advance. That is, the target pressure could just be held at some "negative" amount (not actually negative pressure, since air enters through the nozzle orifice; rather just an equivalent retraction) during non-extrusion/travel moves, instead of at 0.

This should yield significantly faster printing than having the slicer perform retraction, by doing in concurrently with motion, and should also improve print quality/reduce loss of material to oozing by being able to perform retractions in places where slicers don't want to (concurrent with "combing" over already printed lines or infill area, travel between infill lines, etc.) because it would be prohibitively slow to stop and retract.

Nominally, Linear Advance is supposed to reduce or eliminate the need for retraction anyway, but it doesn't, especially with flex materials where the model's accuracy to real material behavior is only so-so, and where the softness of the material has it leaking out even without any pressure.

zillarob, 2019-08-25:
Something to consider is that retract probably doesnt have much effect on the hot goo in the nozzle, beyond relieving pressure.
I think the make-up air comes from the extruder end and through the bowden, not the nozzle.
The solid filament pulls out just fine, but leaves the hot goo in the nozzle.
Really not much you can do about this hot goo oozing, other than move faster than it can ooze.

I did draw up a gizmo to see if sealing the bowden would help create the neg psi we want.
It was noticeable on the spikey retract test part, but pretty minimal.
I was using 1.75mm filament and suspect the small area of what would be the piston just didnt do a whole lot. Seemed like I would have to have an excessive amount of retract to create enough displacement to make a meaningful difference.
Larger 3mm filament might have had more of an effect with its larger area/displacement.

https://www.thingiverse.com/thing:2470621

richfelker, 2019-08-26:
Retraction most certainly does have a major effect beyond relief of pressure. With flex materials, even with linear advance, I need well over 6mm to stop stringing, but increasing it does *completely* stop the stringing. It might not be via air pressure from the other end; more likely, it's a matter of the molten material still having internal bonding, so that the tension of the filament having been pulled back counteracts gravity and bonding with already-extruded material. Air pressure might help when the retraction is rapid, though, and might contribute more when a very narrow bowden is used (I believe some now have 1.9mm diameter or less for 1.75mm filament, whereas older ones were 2mm or larger).

I'm not sure what you're considering an excessive amount. I consider 5-7mm normal, not at all excessive, for a bowden, and 10-15mm well within the "non-exessive" range for flex materials.

Bumpfish, 2019-11-11:
I think this is an interesting idea, to allow retraction (less than 0 speed).   In the picture below, I was underextruding on purpose to see why I was getting so much corner and direction change overextrusion.  I keep adjusting the JUNCTION DEVIATION settings, XY and E acceleration, but cannot get rid of the problem on the line changes (layer in picture is a bottom/top with pattern set to Line in Cura).  I cannot find a setting in Cura to retract in corners or line changes, and I am clearly getting too much nozzle pressure as I enter a change in direction.
[image]

richfelker, 2019-11-11:
@Bumpfish: Setting `retraction_min_travel` (Cura) to 0 and using `lines` pattern (vs `zigzag`) should retract at each direction switch. You *might* also have to disable combing; I don't recall.

Bumpfish, 2019-11-12:
So, I have combing disable and pattern set to lines.  Couldn't find the retraction_min_travel.  I looked at the setting visibility and have it check, but there is a message that says "this setting has been hidden by the active machine and will not be visible".  Turns out, this is a current bug in Cura:  https://github.com/Ultimaker/Cura/issues/6513

richfelker, 2020-01-25:
If I'd like to start experimenting with this idea, can anyone point me to where it should be done? Conceptually I think it's just adding `extrusion>0 ? constant : 0` to the effective advance that linear advance computes (equivalent to subtracting the same constant when not extruding since ultmately extruder position is all relative) to apply during extrusion moves (which might just be `constant` if the expression isn't used in computing extruder position when not extruding). So, aside from adding an extra parameter to `M900` to set the constant (default 0 for no auto-retraction) it seems like it could be a one-line or small-number-of-lines change.

Can someone familiar with the code comment on whether this sounds right and if so, where to find the points to change?

richfelker, 2020-01-25:
It seems it might be somewhat more work to implement than that, in that the start and stop code paths now probably assume the "0 advance" before/after the extrusion is what they need to interpolate from/to, and the extrusion move would start off/end up with filament badly retracted if the offset were just applied like this. So it's likely necessary to really split this off as a separate step.

thierryzoller, 2020-06-12:
any luck?