MarlinFirmware/Marlin #15008: Print head vibrate with MKS Sgen LPC1769

Kyosuke33, 2019-08-20:
Hello, i try to put my problem here after trying all facebook group, forums, sellers...

So, i buy a MKS Sgen, with 3*TMC2208 and 1 LV8729 for my Anycubic Kossel Delta Linear Plus.
I build a Marlin 2.0, LPC1769/MKS Sgen, all compile good without errors.
When i use auto home, all 3 axis goes up great and auto home correctly, but when i use move axis, the print head shake/vibrate a lot when she move.

https://www.youtube.com/watch?v=fqzC6viUyBM

So i switch to a BigTreetech SKR 1.3, with SAME drivers, SAME Marlin, just switch LPC1769 to LPC1768 and change MKS Sgen to BigTreetech SKR 1.3.
ALL work great, auto home, move axis, all the movements are smooth.

After that i switch to a Melzi board, SAME drivers and Marlin 1.1.9
ALL work great, auto home, move axis, all the movements are smooth.

I try to us my MKS Sgen with LPC1768 instead of LPC1769, and MKS Sbase instead of MKS Sgen, but same problem.

I contact MKS ans they send me and other board, but same problem.

I have no idea about the problem. I think Marlin 2.0 don't like the LPC1769 chipset :/

If someone can help me it would be great. thank you.

[Nouveau dossier.zip](https://github.com/MarlinFirmware/Marlin/files/3521818/Nouveau.dossier.zip)

AnHardt, 2019-08-20:
Try without `JUNCTION_DEVIATION`.

Kyosuke33, 2019-08-20:
Oh sorry, i forget to say i try without Junction_Deviation, without linear advance, without s-curve acceleration.
With the skr 1.3 it work with and without all of this option, but with the Sgen the problem is the same with and without.

Kyosuke33, 2019-08-20:
I set my step jumper to 16 (jumper 1 and 2 for my tmc2208). But the Sgen Marlin configuration is based on the Sbase setting, and Sbase doesn't have Jumpers. Maybe there is a conflict somewhere. I'm desesperate to use my Sgen one day...

gloomyandy, 2019-08-21:
Have you tried setting: 
#define MINIMUM_STEPPER_PULSE 2
in Configuration_Adv.h?

Kyosuke33, 2019-08-21:
Don't try this. I'll try that after work. But why would it work with same drivers on the skr ? It's weird. I try and i tell you if it work. Thank you.

gloomyandy, 2019-08-22:
@Kyosuke33 I'm not saying it will fix your problem, but the default pulse timing is basically to do things as fast as possible, the SGen has a faster processor than the SKR and so will generate shorter pulses which may cause problems for the drivers. The above define will use a fixed period for the pulse for both boards (and is the recommended setting for the SKR board as well).

Kyosuke33, 2019-08-22:
Ok, i'll try this today. I did not have time to do it yesterday. Thank you.

Kyosuke33, 2019-08-22:
Re ! the line was not activate, so i remove //? But doesn't work :'(
[image]

Kyosuke33, 2019-08-22:
It's weird than it work great on vertical travel when it home, but a lot of vibration when move horizontally.

Kyosuke33, 2019-08-22:
Oh ! it work ! I have try all the solution but because i'm desesperate, i decide to re-try all the solutions people propose to me, and it work without the Junction deviation... But last time i try this i won't work. So maybe it's a combination with an other thing i change, but don't know what xD
So i can't use Junction Deviation ? Or there is something to set for this to work whith ?
Thank you !

Kyosuke33, 2019-08-22:
I'm lost, now it won't work... don't know why...

Kyosuke33, 2019-08-22:
Ouf ! Just reset Eeprom and it work again^^

AnHardt, 2019-08-22:
Please read #14978 again. Completely. Without further modification Junction Deviation can't work on DELTA machines - if at all. Currently you don't get  Junction Deviation but normal jerk calculations with messed up jerk speeds on kinematic machines when you try to activate it.

Kyosuke33, 2019-08-22:
Ok i read this thank you.

The weird thing is that ok, Junction_Deviation can't be activate on Delta, that's why my Sgen won't work.
But why my Delta work great on Skr 1.3 with Junction_Deviation activate ?

Knifa, 2019-08-22:
> Ok i read this thank you.
> 
> The weird thing is that ok, Junction_Deviation can't be activate on Delta, that's why my Sgen won't work.
> But why my Delta work great on Skr 1.3 with Junction_Deviation activate ?

Activating `JUNCTION_DEVIATION` on Delta printers doesn't actually enable JD except for the extruder. It seems like it is enabled but it isn't really --- some warnings need added. :)

The bug in #14978 causes the default jerk to be set incorrectly recently which causes this weird vibration issue.

Kyosuke33, 2019-08-22:
Yes i just saw your post. But i don't know why it only do this with my Sgen, it's weird.
With my Skr 1.3, Marlin 2.0 with JD, all move work great.
With my Sgeb, exactly the same Marlin 2.0, move slow a vibrating on xy.

Kyosuke33, 2019-08-22:
For my test, i only connect the power supply on the board, the drivers, the endstops and the screen.
So there is 3 differences between Skr 1.3 and Mks Sgen :
- Environment (LPC1768/LPC1769)
- Board pinout (Skr 1.3/Mks Sgen)
- Screen connections (câble are reverse position on the MKS Sgen) 

With this, it's hard to find why it work with one with JD, and need to remove JD for the other one.

AnHardt, 2019-08-22:
Likely the difference is in the one having proper jerk values in EEPROM.

Knifa, 2019-08-22:
> Likely the difference is in the one having proper jerk values in EEPROM.

Yes, I'd be curious to know if it still works fine with the same (as much as possible) configuration and build **and** a clean EEPROM.

Is `0` jerk a supported value at all or is it expected that it would go funny? I'm curious if `0` jerk is maybe exacerbating issues caused by #14532 which @Kyosuke33 is almost definitely going to run into here. I too have an Anycubic Kossel. :smile:

Kyosuke33, 2019-08-23:
It work on the Skr 1.3 with JD and clean Eeprom. It's weird.

boelle, 2019-09-24:
@Kyosuke33 so maybe you just need to clean the Eeprom on MKS Sgen LPC1769 ?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle

Kyosuke33, 2020-01-26:
Sorry, yes need to desactivate Junction Deviation and Clean Eeprom.