MarlinFirmware/Marlin #15011: firmware build location !?

biskero, 2019-08-20:
I just downloaded and build Marlin 2.0.x for the SKR Pro v1.1 but I noticed that the platformio.ini does not contain anymore the directories where the firmware is build and located.

Which directory has the firmware.bin ?

Maybe there should be a comment where is located.
thx

sl1pkn07, 2019-08-20:
.pio

shitcreek, 2019-08-21:
We're not a support group
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-08-21 by shitcreek

biskero, 2019-08-21:
If you read the all post you would check that there is no information on where the .bin file is located after a build, maybe you should put it somewhere in the platformio.ini file so that people know exactly where to look, it's a documentation bug!

AnHardt, 2019-08-21:
The .bin file is currently produced in <*project directory*>/.pio/build/<*platform*>/ .
That's PlatformIO default. We don't force the output to somewhere anymore.
For more information ask PlatformIO.

If you are looking for information how to search for a '*.bin' file in a directory tree, please look up your operating systems file browser search functions documentation.

If you can't see directories/files beginning with a '.', please consult your operating systems manual (hidden directories/files).