MarlinFirmware/Marlin #15012: TMC2209 UART connection error

salacpavel, 2019-08-20:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

antibyte, 2019-08-23:
I am facing the same errors, tried with TMC2208 and TMC2209 on SKR1.1
On the Scope i  can see data beeing sent to the drivers, but LCD says connection error.

antibyte, 2019-08-23:
Installed Klipper on the same Board, 2 x TMC2209 + 1X TMC2208 on a single wire working without problems. So it is a bug in Marlin.

GizmoTheGreen, 2019-08-24:
afaik SKR Mini baords and btt boards with stm32 mcus need BTT version of TMCStepper.
what exactly it changed I don't know but I know it works on my SKR Mini e3 DIP when the normal one does not... https://github.com/bigtreetech/TMCStepper/
also to compile with that lib you need to ass -DHAVE_SW_SERIAL to build flags.

ManuelMcLure, 2019-08-25:
Hmmm... I’m having a similar problem on RAMPS/Re-ARM but since this is the first time I’ve used TMC drivers I don’t know if it’s a bug or something I’m doing wrong.

ManuelMcLure, 2019-08-25:
My problem seemed to involve using 1-wire. It didn't work for me at all - I had to use a Y lead and two separate pins on the board for the connection to succeed.

TheZoker, 2019-08-26:
+1

If I replace TMCSteppers with the one from BigTreeTech, I get an error and the build fails:
```
collect2: error: ld returned 1 exit status
*** [.pio/build/BIGTREE SKR MINI/firmware.elf] Error 1
```

GizmoTheGreen, 2019-08-26:
> +1
> 
> If I replace TMCSteppers with the one from BigTreeTech, I get an error and the build fails:
> 
> ```
> collect2: error: ld returned 1 exit status
> *** [.pio/build/BIGTREE SKR MINI/firmware.elf] Error 1
> ```

did you also add -DHAVE_SW_SERIAL to build flags? this seems to be a must if you have bigtreetechs tmcstepper lib

TheZoker, 2019-08-27:
No with the flag it worked, thank you :)

matheusmbar, 2019-08-28:
The TMCStepper library from BTT implements the software serial library methods. This is required to comunicar with the TMC uart drivers I  their boards. 
The HAL STM32F1 in Marlin firmware implements dummy methods for this library. 

- Using original Marlin firmware without replacing the TMCSTEPPER library will result in connection error due to the dummy software serial methods
- Using BTT TMCStepper library in original Marlin firmware results in build errors due to redefining Software Serial library methods  

Adding  -DHAVE_SW_SERIAL to build flags as stated earlier in this issue ignores the Dummy software serial library from Marlin so the build works and the TMC Connection error goes away. 

Should Marlin implement the real software serial library for STM32F1 ?

PS: my test case is running SKR E3 DIP with tmc2208s

lightface79, 2019-08-28:
Same error on SKR 1.3   lcd says tmc connection error
If i do a M122    It reads the 2208 drivers   but get X	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW

only have 1 2208 driver in my test bench

lightface79, 2019-08-28:
What is the evolution?

Grogyan, 2019-08-28:
I followed the same steps.
Get the same errors

Grogyan, 2019-08-29:
@lightface79 Board needs to be powered externally not from USB

Grogyan, 2019-08-29:
I don't know
There are the same issues raised on [teemuatlut's GitHub](https://github.com/teemuatlut/TMCStepper/issues)

As well as on the Discord channel

samnangor, 2019-08-29:
SKR1.3 + 2209
Same problem 0x00:00:00:00	Bad response!

samnangor, 2019-08-29:
SKR1.3 + 2209
Same problem 0x00:00:00:00	Bad response!
with 24V external power supply.

Grogyan, 2019-08-29:
@samnangor Please provide configs.
I tested the SKR 1.3 + 2209 last night with the Marlin Bug Fix from that time

samnangor, 2019-08-29:
Here is my config
Thanks for your help.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3557682/Marlin.zip)

Grogyan, 2019-08-29:
ZIP is invalid

samnangor, 2019-08-29:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3557691/Configuration.zip)
Here is a compress file

Grogyan, 2019-08-29:
Configuration_ADV.h is needed too

samnangor, 2019-08-29:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3557706/Configuration.zip)
This one include 
Configuration.h
Configuration_adv.h

Grogyan, 2019-08-30:
#define SERIAL_PORT 0
Should be -1

samnangor, 2019-08-30:
Thanks Grogyan,
How about SERIAL_PORT 2?

samnangor, 2019-08-30:
Compile with 
SERIAL_PORT_0 -1
Run Pronterface command M122 
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!

Grogyan, 2019-08-30:
Watterot or BTT 2209?

samnangor, 2019-08-30:
This is what it prints on the PCB
ACCA 1929
FYSETC               Silent 2209
               V2.1

Grogyan, 2019-08-30:
Make sure the PDN UART jumper is soldered

samnangor, 2019-08-30:
I am still looking for PDN UART jumper location on my TMC2209 PCB
Thanks again Grogyan.

samnangor, 2019-08-30:
Found it, I have to short between Tx and Rx

Erik9519, 2019-09-07:
I've just received my SKR 1.3 board and 4 GEEETECH TMC2208 and I'm also getting "Bad response" on all 4. 
I have soldered the jumper pad for the UART on the drivers, set all the jumpers correctly on the SKR and the drivers do work as I can manually move them via the LCD.
How to fix this?

lightface79, 2019-09-07:
Erik9519
I solved my problem om ny stepper errors.
Need to power the board externaly from 12 or 24v   so no USB powereing the board.  Set the dip to ext power. 

Second thing That was wrong was i had 3 pads on my 2208 and u should Only have the middle pad and one or the other left or right pads solderd together with the middle one.   I had mine solderd go the wrong pad.   So the uart was in the wrong pin to the skr1.3

lightface79, 2019-09-07:
So when i changed the pads the uart was working perfect.

lightface79, 2019-09-07:
Then came the second issue.
Since my drivers worked like yours in standalone mode.

When the uart mode worked i could not move my steppers..  

Found out that i needed to disabel software_enable

Then everything worked like a charm

Erik9519, 2019-09-08:
lightface79
After some extensive debugging over the Marlin Discord we came to the conclusion that to solve my problem I had to solder together the unmarked pin on my drivers with the PDN pin on top of soldering the uart pads.
Once I did that they worked just fine in UART mode.

These were the drivers: GEEETECH TMC2208
[https://www.amazon.de/GEEETECH-TMC2208-Stepstick-Heatsink-Screwdriver/dp/B07MZTCSZY](url)

boelle, 2019-09-24:
@salacpavel still having problems?

samnangor, 2019-09-25:
Not the same problem though 
I  cannot make my extruder fan come on automatically It should come on when extruder temperature goes above 50 I have tried to raise the temperature to 80 and it still won't run.  It will turn on and off manually when I send Gcode command.
Attached is the file
Thanks

TheZoker, 2019-09-25:
So I updated to the [latest commit](https://github.com/MarlinFirmware/Marlin/commit/c3f09e2b1bde37bfec8e7de877571455d750c64c) today and get this issue with the bigtreetech TMCStepper library:
```
In file included from Marlin\src\module/stepper/indirection.h:44:0,
                 from Marlin\src\module/stepper.h:47,
                 from Marlin\src\Marlin.cpp:37:
Marlin\src\module/stepper/trinamic.h:31:4: error: #error "Update TMCStepper library to 0.5.0 or newer."
   #error "Update TMCStepper library to 0.5.0 or newer."
    ^~~~~
Compiling .pio\build\STM32F103R_bigtree\src\src\feature\bedlevel\abl\abl.cpp.o
In file included from Marlin\src\feature\../module/stepper/indirection.h:44:0,
                 from Marlin\src\feature\../module/stepper.h:47,
                 from Marlin\src\feature\babystep.cpp:30:
Marlin\src\feature\../module/stepper/trinamic.h:31:4: error: #error "Update TMCStepper library to 0.5.0 or newer."
   #error "Update TMCStepper library to 0.5.0 or newer."
    ^~~~~
Compiling .pio\build\STM32F103R_bigtree\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\STM32F103R_bigtree\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o
Compiling .pio\build\STM32F103R_bigtree\src\src\feature\bedlevel\ubl\ubl.cpp.o
Compiling .pio\build\STM32F103R_bigtree\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o
*** [.pio\build\STM32F103R_bigtree\src\src\Marlin.cpp.o] Error 1
*** [.pio\build\STM32F103R_bigtree\src\src\feature\babystep.cpp.o] Error 1
```

Does anyone got a clue about that?

boelle, 2019-09-25:
> Not the same problem though  I  cannot make my extruder fan come on automatically It should come on when extruder temperature goes above 50 I have tried to raise the temperature to 80 and it still won't run.  It will turn on and off manually when I send Gcode command. Attached is the file Thanks

you can not attach files from email, you have to use github website

boelle, 2019-09-25:
> So I updated to the [latest commit](https://github.com/MarlinFirmware/Marlin/commit/c3f09e2b1bde37bfec8e7de877571455d750c64c) today and get this issue with the bigtreetech TMCStepper library:
>
> ``` [...]

look for this in your error: Update TMCStepper library to 0.5.0 or newer

it might give you a clue :-D

TheZoker, 2019-09-25:
@boelle But are the issues with the original TMCStepper library fixed?

boelle, 2019-09-25:
i have no clue, that is why i ask

personally i use DRV8825 and TMC2100 and those dont require drivers at all

TheZoker, 2019-09-25:
I changed the library back to the original library and uploaded the firmware to my ender 3 and the `TMC connection error` is still present. 

So now the issue is:
- I can't use the [original TMCStepper Library](https://github.com/teemuatlut/TMCStepper), because it can't talk to the TMC 2209 
- I can't use the [BIGTreeTech TMCStepper](https://github.com/bigtreetech/TMCStepper) Library, because the code won't compile with the error above

boelle, 2019-09-25:
and you did update to version 0.5.0 ?

TheZoker, 2019-09-25:
I used the [original library annotation](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/platformio.ini#L30) and from the `.pio/libdeps/TMCStepper` folder I can see, that pio installed version 0.5.0:
[image]

boelle, 2019-09-25:
very strange that says to update to 0.5.0 :-/

TheZoker, 2019-09-25:
@thinkyhead Any idea about this error?

randellhodges, 2019-09-25:
@TheZoker There is 2 TMCStepper folders in your screenshot. Maybe removed the one without a suffix, or the entire libdeps folder and let it download fresh again?

TheZoker, 2019-09-25:
Just tried that (removing the whole folder and let it download all the libs again) but same, TMC connection error:
[image]

This seems to be the explanation of the TMC connection issue:
https://github.com/MarlinFirmware/Marlin/issues/15012#issuecomment-525572365

It was possible to work around this by using the bigtreetech TMCStepper library, but yeah, that one does not compile anymore...

randellhodges, 2019-09-25:
Oh sorry, I didn't realize you were talking about a runtime error, I thought it was the compile error people were getting. Good to know because I have a SKR 1.3 that I was getting ready to put in my 2209 steppers. I have the firmware compiled, just haven't gotten around to updating yet. I guess I'll wait :)

TheZoker, 2019-09-25:
The issue seems to be this commit: 
https://github.com/MarlinFirmware/Marlin/commit/bd2b44c4ed4221d30f3bceac90236632fe6a6411

sjasonsmith, 2019-09-25:
Marlin has an unimplemented SoftwareSerial library that lets you compile, but doesn't let anything work. It seems it would be much better for it to break the build than fail at runtime.

Try adding the FYSETC SoftwareSerialM library to your STM32F103R_bigtree to see if it works. I got mine working yesterday using that. Here is the entire block you can copy/paste to see if it works.

```
#
# BigTree SKR Mini V1.1 / SKR mini E3 / SKR E3 DIP (STM32F103RCT6 ARM Cortex-M3)
#
[env:STM32F103R_bigtree]
platform          = ststm32
framework         = arduino
board             = genericSTM32F103RC
platform_packages = tool-stm32duino
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F1_SKR_MINI.py
build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py
  ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DHAVE_SW_SERIAL
build_unflags     = -std=gnu++11
lib_deps          = ${common.lib_deps}
  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
lib_ignore        = Adafruit NeoPixel, SPI
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
monitor_speed     = 115200
upload_protocol   = stlink
debug_tool        = stlink```

sjasonsmith, 2019-09-25:
The actual changes in my last post were the `lib_deps` line, and adding `-DHAVE_SW_SERIAL` to the `build_flags`.

TheZoker, 2019-09-26:
@sjasonsmith I tried that, but got the same TMC connection error again :/

samnangor, 2019-09-26:
Try solder the Tx and Rx pin (on the TMC2209 board) together.
Mine is working fine. Just did my first print.

sjasonsmith, 2019-09-26:
I have it working with an **SKR E3 Mini (Built-in 2209 drivers)** and an **SKR E3 DIP (2208 Drivers)**.

I do not know if The E3 DIP board needs anything special to work with 2209 drivers. My DIP config compiles fine with the drivers changed to TMC2209, but I don't have the hardware to test it. Perhaps jumpers on the board can change the slave addresses and require modifying the _SLAVE_ADDRESS settings in configuration_adv.h.

I can't speak for SKR Mini 1.1 (non-E3) boards at all. It looks like the pins file does not define any serial pins, so much more work would be needed to configure it properly.

I won't have access to my machines for the rest of this week, so I'll post my working configs for these right now, in case that helps anybody.

[SKR_E3_DIP.zip](https://github.com/MarlinFirmware/Marlin/files/3659554/SKR_E3_DIP.zip) (2208 UART drivers. Never tested in an actual printer, might have other config issues)
[SKR_E3_Mini.zip](https://github.com/MarlinFirmware/Marlin/files/3659555/SKR_E3_Mini.zip) (Built-in 2209 Drivers. From an Ender 5 with BLTouch, but not fully tested)

TheZoker, 2019-10-03:
Any news on this issue?
Why is the SoftwareSerial library not added in Marlin?

TheFoXi, 2019-10-08:
I think need to add an extra build_flags to compile  **-DTMC_SERIAL_SWITCH**
It helped me for E3 DIP and 2208 UART.
(https://github.com/teemuatlut/TMCStepper/blob/b5b3658d34f93e0cc646dbaa54f90b809267206f/src/TMCStepper.h#L26)

LastDragon-ru, 2019-10-09:
56595a4c9cfad38a5fdd46d48e370edf70e0a079 + Skr 1.3 + 2208 (or 2209) = uart is totally broken :( 

> -DTMC_SERIAL_SWITCH

```
Compiling .pio\build\LPC1768\libffb\TMCStepper_ID5513\source\SHORT_CONF.cpp.o
.pio\libdeps\LPC1768\TMCStepper_ID5513\src\source\SERIAL_SWITCH.cpp: In member function 'void SSwitch::active()':
.pio\libdeps\LPC1768\TMCStepper_ID5513\src\source\SERIAL_SWITCH.cpp:16:7: error: 'writePIN_L' was not declared in this scope
       writePIN_L(p1);
       ^~~~~~~~~~
.pio\libdeps\LPC1768\TMCStepper_ID5513\src\source\SERIAL_SWITCH.cpp:16:7: note: suggested alternative: 'writeSCK_L'
       writePIN_L(p1);
       ^~~~~~~~~~
       writeSCK_L
.pio\libdeps\LPC1768\TMCStepper_ID5513\src\source\SERIAL_SWITCH.cpp:21:7: error: 'writePIN_H' was not declared in this scope
       writePIN_H(p1);
       ^~~~~~~~~~
.pio\libdeps\LPC1768\TMCStepper_ID5513\src\source\SERIAL_SWITCH.cpp:21:7: note: suggested alternative: 'writeSCK_H'
       writePIN_H(p1);
       ^~~~~~~~~~
       writeSCK_H
Compiling .pio\build\LPC1768\libffb\TMCStepper_ID5513\source\SMARTEN.cpp.o
*** [.pio\build\LPC1768\libffb\TMCStepper_ID5513\source\SERIAL_SWITCH.cpp.o] Error 1
```

randellhodges, 2019-10-09:
I'm compiling with SKR 1.3 + 2208 with commit e6055dc. Using TMCStepper library 0.5.1 (I haven't changed any pio configuration options and not using bigtree tmcstepper fork. I haven't printed yet, but it moves all the axis and homes fine.

[24 more replies]

closed 2019-10-12 by boelle