MarlinFirmware/Marlin #15014: [FR] Support E1 as AXIS for new switching extruder

kerbetos, 2019-08-21:
good day
I turn to the following question: I made an extruder with a change of material, the extruder must use only two motors, one is connected to E0 and provides plastic supply, the second motor of the extruder E1 must provide plastic change
The device will use an optical limit switch to determine the start of the selection of the extruder.
On the block will be involved any number of extruders from 3 pieces or more.
in Configuration.h we define
#define EXTRUDERS 4
#define SINGLENOZZLE
For further work, you need to implement the ability to control the driver of the stepper motor E1 as a normal axis (like the XYZ axis) with the ability to park on a sensor connected to an unused one, for example X_MAX
The rest of the extruder control will already be through the G-code

kerbetos, 2019-08-21:
add photos of the device
[https://photos.app.goo.gl/JDhMMeJBGpey9WA36](https://photos.app.goo.gl/JDhMMeJBGpey9WA36)

AnHardt, 2019-08-21:
Marlin has 4 **independent** axes!
One for X, one for Y, one for Z and one shared by the extruders!
This will not change soon - if ever!

In systems like Prusas MMU2 a separate processor is used.

kerbetos, 2019-08-21:
not yet, but I think things will change soon

zillarob, 2019-08-22:
I wonder if you could cheat it for now by setting endstops to default always on and messing with what it does on endstop hit?
If you get creative here you might be able to get it working well enough to play with it for now.

Developkings, 2019-10-14:
can,t you create a custom code to add to the extruder change so that you use absolute coordinates in the second extruder to define the position of each filament and then redefine the extruder one?

Im going to try to ruffly describe what i mean, not writing anything close to code loool
/*
int X = defined_extruder
int A = 0
int B = 20
int C = 40

switch (x) {
case 0:
    E1 = Go to absolute position A
   defineActiveExtruder: 0
break;

case 1:
    E1 = Go to absolute position b
   defineActiveExtruder: 0
break;

case 0:
    E1 = Go to absolute position c
   defineActiveExtruder: 0
break;
 }
Resume Print
*/

after that it would resume the print normally not being aware that it was tricked to change the extruder

thisiskeithb, 2021-09-24:
Does the 6-axis support added in 2.0.9 work for this request?

DerAndere1, 2022-02-20:
Marlin has support for up to 6 axes plus extruders and also the option that could be repurposed to drive a tool changer. `EVENT_GCODE_AFTER_TOOLCHANGE`, `EVENT_GCODE_TOOLCHANGE_T0`, `EVENT_GCODE_TOOLCHANGE_T1`, `EVENT_GCODE_TOOLCHANGE_ALWAYS_RUN`. If I understand correctly, issue https://github.com/MarlinFirmware/Marlin/issues/23003 is related

closed 2026-05-09 by thinkyhead