MarlinFirmware/Marlin #15022: BUG! Pronterface not able to writefile, the device does not recog

Jchisholm204, 2019-08-21:
-->

### Description

After uploading marlin bugfix 2.0 my board was unable to connect to my win10 pc or octoprint.  The board shows up, then goes away a few seconds later in device manager.

Error Message from Pronterface:
Connecting...
Printer is now online.
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
[ERROR] Can't write to printer (disconnected?) (SerialException): WriteFile failed (WindowsError(22, 'The device does not recognize the command.'))

### Steps to Reproduce

1. upload marlin bugfix 2.0 from bigtree tech to skr v1.3 mainboard
2. restart board
3. plug into pc.. no longer works

**Expected behavior:** 

board recognised by pc, able to use over serial port
**Actual behavior:** 
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3527698/Marlin-bugfix-2.0.x.zip)
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/3527699/Marlin-bugfix-2.0.x.zip)

Error given and board appears to reset after a few seconds
#### Additional Information
I have included my entire code.. Any help appreciated, I am new to marlin and have never flashed a board before.  I am worried I screwed up the bootloader, if I did, how do I fix that?

DerAndere1, 2019-08-23:
This could be a problem with your wire connections. In general it is recommended to talk to experienced Marlin users before submitting a bug report (e.g. on forums like the [Marlin facebook group](https://www.facebook.com/groups/1049718498464482/) or the [reprap - Marlin forum](https://reprap.org/forum/list.php?415)).  
Still, here my thoughts: First verify that the baudrate in your configuration.h is set to a value that is proven to work with your board. Specify the same baudrate in the host software (pronterface) for connection. When I google the error messages you obtained, I found the following tips that may help you: https://reprap.org/forum/read.php?4,106212,106622,quote=1 and https://3dprinting.stackexchange.com/questions/5270/serialexception-writefile-failed-error-22-printrbot-simple-metal-using. Please give feedback if any of the above fixed your problem.

Jchisholm204, 2019-08-23:
@DerAndere1 thanks for your support, it turned out to be a problem with the bootloader which I was able to fix by flashing smoothie through the SD card.  I am trying to flash marlin again today, possibly without breaking the bootloader

thisiskeithb, 2019-08-24:
@Jchisholm204: You mention it being a bootloader, but did you have thermistors plugged into the board? If not, Marlin will halt/disconnect from your host shortly after booting.

Jchisholm204, 2019-08-24:
@thisiskeithb Yes, I had everything plugged into the board

Jchisholm204, 2019-08-24:
everything is now working fine under marlin, so thankyou and I now have an octoprint error to deal with

closed 2019-08-24 by Jchisholm204