MarlinFirmware/Marlin #15028: [BUG] Error compiling tmc2209 LPC1768

daverdfw, 2019-08-22:
Attempting to compile 2.0bugfix 13d8dc0b for TMC2209 support, previously had TMC2130 compiled without issue and receiving the following error

Marlin/src/module/endstops.cpp: In static member function 'static bool Endstops::tmc_spi_homing_check()':
Marlin/src/module/endstops.cpp:797:40: error: 'class TMCMarlin<TMC2209Stepper, 'X', '0', (AxisEnum)0>' has no member named 'test_stall_status'; did you mean 'get_stealthChop_status'?
       if (tmc_spi_homing.x && stepperX.test_stall_status()) {
                                        ^~~~~~~~~~~~~~~~~
                                        get_stealthChop_status
Marlin/src/module/endstops.cpp:803:40: error: 'class TMCMarlin<TMC2209Stepper, 'Y', '0', (AxisEnum)1>' has no member named 'test_stall_status'; did you mean 'get_stealthChop_status'?
       if (tmc_spi_homing.y && stepperY.test_stall_status()) {

daverdfw, 2019-08-22:
looks like this was all changed with this commit, so perhaps something else is broken ?

https://github.com/MarlinFirmware/Marlin/commit/d4974ea719ceba2d76a678f77942d0b92e8fecb8#diff-f3f16ada73f2765518ee060b71e3de26

teemuatlut, 2019-08-22:
https://github.com/MarlinFirmware/Marlin/blob/e604f76703165027a22dc33c7875f4858e9f1c80/Marlin/Configuration_adv.h#L1924

You're trying to use a feature that's not yet supported for 2209.

daverdfw, 2019-08-22:
> 

> https://github.com/MarlinFirmware/Marlin/blob/e604f76703165027a22dc33c7875f4858e9f1c80/Marlin/Configuration_adv.h#L1924
> 
> You're trying to use a feature that's not yet supported for 2209.

well that explains it then! so no sensorless homing for the 2209 yet?

ManuelMcLure, 2019-08-22:
You can do sensorless homing with a wire from the DIAG pin to the endstop connector - it's the SPI "no extra wires" homing that isn't supported.

daverdfw, 2019-08-22:
> You can do sensorless homing with a wire from the DIAG pin to the endstop connector - it's the SPI "no extra wires" homing that isn't supported.

I am using a SKR 1.3 which has the pin on the board for DIAG, so that should work just fine, correct?

teemuatlut, 2019-08-22:
I never tested it when adding the 2209. The intention was that I'll first do the general support so people can use the new drivers and then later come back to address sensorless homing. But it turned out that we had reports of the Sensorless Homing working right away and so I had do issues leaving it listed.

And yea, like Manuel said, the SPI homing is another feature that adds to sensorless homing.
2209 support will come at some point when I have the time but atm I've got quite a lot going on.

With SKR v1.3 you'd want to stick with normal wired mode as it is already wired in.

daverdfw, 2019-08-22:
> I never tested it when adding the 2209. The intention was that I'll first do the general support so people can use the new drivers and then later come back to address sensorless homing. But it [...]

perfect, thanks so much for the response, I didn't catch I even had the SPI homing enabled, do you know on the SKR if I have to use SW SPI? or will HW work on this board?

teemuatlut, 2019-08-22:
The SKR v1.3 is designed for SW SPI.

daverdfw, 2019-08-22:
> The SKR v1.3 is designed for SW SPI.

that's what I thought because it doesn't have as many pins like the MKS Gen L did for the HW connections, I will close out this issue, thanks!

closed 2019-08-22 by daverdfw

petermccracken, 2020-05-06:
What happens if you use 4 x TMC2209 and 1 x TMC5160. You can't enable the SPI ENDSTOPS feature to use sensorless homing on the 5160 as the TMC2209 codes throws this error. Surely this is a bug as the 2209 code should ignore this flag whist the 2130/5160 code should implement sensorless homing via SPI ? Thanks!