MarlinFirmware/Marlin #15029: TMC2208 COMPILE ERROR

SISLANGER, 2019-08-22:
Cant compile after latest update, I was having problems saving ABL data via external EEPROM on SKR PRO so thought I would try firmware update today from Marlin GitHub. However below are the errors I am receiving todo with TMC2208.

In file included from Marlin/src/module/stepper_indirection.h:48:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
Marlin/src/module/../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2208Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'X'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)0; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin/src/module/stepper_indirection.cpp:317:7:   required from here
Marlin/src/module/../feature/tmc_util.h:169:40: error: no matching function for call to 'TMC2208Stepper::TMC2208Stepper(const uint16_t&, const uint16_t&, const float&, const bool&)'
       TMC2208Stepper(RX, TX, RS, has_rx)
                                        ^
In file included from Marlin/src/module/stepper_indirection.h:47:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, uint8_t)
   TMC2208Stepper(Stream * SerialPort, float RS, uint8_t addr);
   ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:802:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, bool)
   TMC2208Stepper(Stream * SerialPort, float RS, bool has_rx=true) :
   ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:802:3: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(const TMC2208Stepper&)
 class TMC2208Stepper : public TMCStepper {
       ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(TMC2208Stepper&&)
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided
In file included from Marlin/src/module/stepper_indirection.h:48:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
Marlin/src/module/../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2208Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'Y'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)1; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin/src/module/stepper_indirection.cpp:331:7:   required from here
Marlin/src/module/../feature/tmc_util.h:169:40: error: no matching function for call to 'TMC2208Stepper::TMC2208Stepper(const uint16_t&, const uint16_t&, const float&, const bool&)'
       TMC2208Stepper(RX, TX, RS, has_rx)
                                        ^
In file included from Marlin/src/module/stepper_indirection.h:47:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, uint8_t)
   TMC2208Stepper(Stream * SerialPort, float RS, uint8_t addr);
   ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:802:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, bool)
   TMC2208Stepper(Stream * SerialPort, float RS, bool has_rx=true) :
   ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:802:3: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(const TMC2208Stepper&)
 class TMC2208Stepper : public TMCStepper {
       ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(TMC2208Stepper&&)
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided
In file included from Marlin/src/module/stepper_indirection.h:48:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
Marlin/src/module/../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2208Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'Z'; char DRIVER_ID = '0'; AxisEnum AXIS_ID = (AxisEnum)2; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin/src/module/stepper_indirection.cpp:345:7:   required from here
Marlin/src/module/../feature/tmc_util.h:169:40: error: no matching function for call to 'TMC2208Stepper::TMC2208Stepper(const uint16_t&, const uint16_t&, const float&, const bool&)'
       TMC2208Stepper(RX, TX, RS, has_rx)
                                        ^
In file included from Marlin/src/module/stepper_indirection.h:47:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, uint8_t)
   TMC2208Stepper(Stream * SerialPort, float RS, uint8_t addr);
   ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:802:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, bool)
   TMC2208Stepper(Stream * SerialPort, float RS, bool has_rx=true) :
   ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:802:3: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(const TMC2208Stepper&)
 class TMC2208Stepper : public TMCStepper {
       ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(TMC2208Stepper&&)
[... 22 lines cut ...]
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(TMC2208Stepper&&)
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided
In file included from Marlin/src/module/stepper_indirection.h:48:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
Marlin/src/module/../feature/tmc_util.h: In instantiation of 'TMCMarlin<TMC2208Stepper, AXIS_LETTER, DRIVER_ID, AXIS_ID>::TMCMarlin(uint16_t, uint16_t, float, uint8_t, bool) [with char AXIS_LETTER = 'E'; char DRIVER_ID = '1'; AxisEnum AXIS_ID = (AxisEnum)4; uint16_t = short unsigned int; uint8_t = unsigned char]':
Marlin/src/module/stepper_indirection.cpp:373:7:   required from here
Marlin/src/module/../feature/tmc_util.h:169:40: error: no matching function for call to 'TMC2208Stepper::TMC2208Stepper(const uint16_t&, const uint16_t&, const float&, const bool&)'
       TMC2208Stepper(RX, TX, RS, has_rx)
                                        ^
In file included from Marlin/src/module/stepper_indirection.h:47:0,
                 from Marlin/src/module/stepper_indirection.cpp:32:
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, uint8_t)
   TMC2208Stepper(Stream * SerialPort, float RS, uint8_t addr);
   ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:966:3: note:   candidate expects 3 arguments, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:802:3: note: candidate: TMC2208Stepper::TMC2208Stepper(Stream*, float, bool)
[... cut ...]
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(const TMC2208Stepper&)
 class TMC2208Stepper : public TMCStepper {
       ^~~~~~~~~~~~~~
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note: candidate: constexpr TMC2208Stepper::TMC2208Stepper(TMC2208Stepper&&)
.pio/libdeps/BIGTREE_SKR_PRO/TMCStepper_ID5513/src/TMCStepper.h:800:7: note:   candidate expects 1 argument, 4 provided

teemuatlut, 2019-08-22:
SKR Pro is an STM32 based board and it doesn't have a SoftwareSerial implementation within the framework. As such it is not yet considered a SW Serial capable platform by the TMCStepper library.

SISLANGER, 2019-08-22:
@teemuatlut thank you, you reminded me of the different TMC library required.

If anyone else forgets or is having same its this one you need to add to platformio.ini https://github.com/bigtreetech/TMCStepper

closed 2019-08-22 by SISLANGER