MarlinFirmware/Marlin #15030: [BUG] Tmc2209 sensitivity inverse help text

lokus77, 2019-08-22:
The documentation (help text)  in "Configuration_adv.h" about  tmc2130/2209 sensitivity is wrong:

"* X/Y/Z_STALL_SENSITIVITY is used to tune the trigger sensitivity.
   * Use M914 X Y Z to live-adjust the sensitivity.
   *  Higher: LESS sensitive. (Too high => failure to trigger)
   *   Lower: MORE sensitive. (Too low  => false positives)
"
This is (maybe) true in tmc2130, but is false in tmc2209.
Correct is opposite:
   *  Lower : LESS sensitive. (Too low => failure to trigger)
   *  Higher: MORE sensitive. (Too high   => false positives)

// TMC2209: 0...255. 
default
    #define X_STALL_SENSITIVITY  8
    #define Y_STALL_SENSITIVITY  8

reloxx13, 2019-08-23:
duplicate https://github.com/MarlinFirmware/Marlin/issues/14921

boelle, 2019-08-23:
if @lokus77 can agree that #14921 is the same thing then i think we can close this one as 10 issues for the same thing just confuses and clutter things more than it does good

lokus77, 2019-08-23:
sorry, duplicated with  #14921 I close it!

maybe @teemuatlut  tmcstepper driver has originated the Wrong description, cause 
"
* A higher STALL_VALUE will make the reading less sensitive and
 * a lower STALL_VALUE will make it more sensitive
"
in examples stallguard.
https://github.com/teemuatlut/TMCStepper/blob/master/examples/StallGuard_TMC2209/StallGuard_TMC2209.ino

closed 2019-08-23 by lokus77