MarlinFirmware/Marlin #15033: [BUG] Compilation errors when TURBO_BACK_MENU_ITEM defined

Patag, 2019-08-23:
### Steps to Reproduce

1. [First Step]
define TURBO_BACK_MENU_ITEM in Configuration_adv.h
2. [Second Step]
Start compiling
**Expected behavior:** [What you expect to happen]
Compilation OK
**Actual behavior:** [What actually happens]
Every call to `ui.goto_previous_screen` function leads to an error such this one :
`Marlin\src\lcd\menu\menu_advanced.cpp:613:16: error: invalid conversion from void (*)(bool) to selectFunc_t {aka void (*)()}' [-fpermissive]
             ui.goto_previous_screen, PSTR(MSG_INIT_EEPROM), nullptr, PSTR("?"));`
             ~~~^~~~~~~~~~~~~~~~~~~~

NovaViper, 2019-08-23:
I can confirm the same thing

Ludy87, 2019-08-23:
Guys, there are a lot of settings, please add the configuration files!

Patag, 2019-08-23:
@Ludy87 please, don't hide the issue behind the lack of conf files !
Try yourself to activate TURBO_BACK_MENU_ITEM, whatever your conf, and you will see the results.
BTW, I still can't understand how is it possible to merge non-compilable code...

Ludy87, 2019-08-23:
Now that I'm bored, I've found out that 'EEPROM_SETTINGS' must be ENABLED.

thank you for your cooperation

Ludy87, 2019-08-23:
See: https://github.com/MarlinFirmware/Marlin/pull/15019

boelle, 2019-08-24:
i assume that https://github.com/MarlinFirmware/Marlin/pull/15019 fix's the issue, we can always reopen this one if that is not the case

closed 2019-08-24 by boelle

Patag, 2019-08-24:
Issue fixed.
Thank you