MarlinFirmware/Marlin #15034: Dual Extruder auto-level problem

davidhk21010, 2019-08-23:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

davidhk21010, 2019-08-23:
Description: Dual extrusion with auto-level causes ghosting and part intersection with second extruder.

[image]

I'm using a gcreate gmax 1.5 dual extruder printer with auto-leveling on. I believe we're using ABL, not UBL.

The auto level works well, and I get good bed adhesion across the first layer. However, two issues come up:

1. At lower layers we get ghosting in some areas.
2. At higher layers we get part intersection that predictably results in layer separation.

After running a number of test prints, I was able to identify the problem. First our bed is VERY level (Max diff from one side to the other of the large bed is 0.35mm Most of the bed has a diff of 0.05mm.

Second, we have run numerous tests to make sure the x-axis is level.

When everything is very level, the auto level causes the second extruder to dip into the bed or the print because the auto-level only accounts for the placement of the head that's printing at that location. It doesn't account for the z-axis height difference at the second extruder. It seems that the easy fix is for the bed leveling map to account for the x-axis placement of the second extruder and set that location as a minimum for the z-drop on the auto-level. Therefore, only in the locations where the second extruder would go lower than the current extruder, the program wouldn't allow the drop. It's almost like two z location checks at any given moment.

The UBL fade function fixes most of this issue, but it would be a much better solution to have the minimum in place and the issue never occurs in the first place. The fix is not ideal because there are some locations where the primary extruder may not go low enough, however, in the dual-extruder circumstance, it would be a trade off.

If you're printing a large enough part, a few small areas that are very slightly too high would not have a serious impact compared to the second extruder impact at higher layers.

InsanityAutomation, 2019-08-28:
Unfortunately this is an intrinsic problem with fixed dual extruders. There is simply no physical way around this, you either don't use mesh bed leveling or use one of the many methods that can physically move the second tool head. The linear methods can have some success if the tilt of the tool head matches the tilt of the bed...

AnHardt, 2019-08-29:
Imagine a flat level bed and different high nozzles. When the higher one is on the surface the deeper one is below. You can print only if the the nozzles are at the same hight.
On a flat tilt bed the solution is to till the nozzles to the exactly same degree. Look at the Kraken nozzle adjustment mechanism/procedure. (Loosen the nozzles, drop them to the bed, tighten then.)
For a bumpy bed no solution exists. You cant go to the deepest point in a valley with one nozzle, without ramming the other into a hillside.

shitcreek, 2019-09-05:
Closing this because this is a physical impossibility.

closed 2019-09-05 by shitcreek