MarlinFirmware/Marlin #15035: [FR] Dynamic Acceleration Adjustment

AmyTheCute, 2019-08-23:
### Description

Dynamic Acceleration control for Marlin!

#### Additional Information

https://duet3d.dozuki.com/Wiki/Gcode#Section_M593_Configure_Dynamic_Acceleration_Adjustment

this is stupidly cool! i saw a few people post the results and it makes a world of difference! it'd be great to have on marlin 2.0 :) specially with the 32-bit boards!

Minims, 2020-12-07:
Is it the same thing as Resonance Compensation ?  :  https://www.klipper3d.org/Resonance_Compensation.html

haarp, 2020-12-26:
I believe this is what #16531 is about. Am I wrong?

thisiskeithb, 2022-05-10:
> I believe this is what #16531 is about. Am I wrong?

Indeed. Please migrate discussion to #16531 since it has had a lot more discussion.

closed 2022-05-10 by thisiskeithb