MarlinFirmware/Marlin #15036: [BUG] Unknown environment names BIGTREE_SKR_MINI

tensau, 2019-08-23:
Marlin does not compile in VS Code / Platformio with the following configuration in platformio.ini:

> [platformio]
> src_dir     = Marlin
> build_dir   = .pioenvs
> lib_dir     = .piolib
> libdeps_dir = .piolibdeps
> env_default = BIGTREE_SKR_MINI
> 

Error message:

> Error: Unknown environment names 'BIGTREE_SKR_MINI'. Valid names are 'melzi, at90USB1286_DFU, teensy20, megaatmega2560, at90USB1286_CDC, sanguino_atmega644p, megaatmega1280, melzi_optiboot, sanguino_atmega1284p, rambo'

RudyBenner, 2019-08-24:
[image]

thisiskeithb, 2019-08-24:
Are you using the latest `bugfix-2.0.x`? `platformio.ini` has been updated and no longer looks like that:

https://github.com/MarlinFirmware/Marlin/blob/d6a16525c2742719ce2f477738dd548f27015328/platformio.ini#L18-L21

tensau, 2019-08-25:
Shame on me, i indeed had the wrong commit :(
Everything works now, thanks for your help!

closed 2019-08-25 by tensau

alexws54tk, 2019-09-04:
Latest `Marlin bugfix-2.0.x` does not compile in VS Code / Platformio with the following configuration in platformio.ini:

> [platformio]
> src_dir = Marlin
> build_dir = .pioenvs
> lib_dir = .piolib
> libdeps_dir = .piolibdeps
> env_default = BIGTREE_SKR_MINI

Error message:

> Error: Unknown environment names '**BIGTREE_SKR_MINI**'. Valid names are 'melzi, STM32F103R_fysetc, teensy31, rambo, at90usb1286_cdc, megaatmega1280, mks_robin, BIGTREE_SKR_PRO, mks_robin_nano, STM32F103V_longer, STM32F103R, STM32F407VE_black, mks_robin_mini, at90usb1286_dfu, sanguino_atmega1284p, STM32F4, STM32F7, ARMED, BIGTREE_BTT002, STM32F103R_bigtree, megaatmega2560, teensy35, DUE, include_tree, jgaurora_a5s_a1, DUE_USB, DUE_debug, SAMD51_grandcentral_m4, esp32, sanguino_atmega644p, malyanm200, LPC1768, LPC1769, linux_native, melzi_optiboot, mks_robin_lite, fysetc_f6_13'

p3p, 2019-09-04:
The `BIGTREE_SKR_MINI` environment was changed to `STM32F103R_bigtree` in a recent cleanup as it applies to more than just the mini.

thisiskeithb, 2019-09-04:
When in doubt, read through Platformio.ini:

https://github.com/MarlinFirmware/Marlin/blob/d9bbe0ff02530239dfa726b87e8303ff427755ba/platformio.ini#L296-L299