MarlinFirmware/Marlin #15039: [FR] CR-10 Example Profile Thermistor Incorrect

Deneteus, 2019-08-24:
### Description: 

The thermistor type for the CR-10 and CR-10S is incorrect in the Example profile.

The thermistor for the heat bed is listed as 100K ohm NTC 3950FB. In the original CR-10 source Creality used 1 for both the bed and the hot end. These are also the only thermistors they sell for this printer.

### Steps to Reproduce

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/example_configurations/Creality/CR-10/Configuration.h

[image]

#### Additional Information

Creality CR-10 Github Config.h
https://github.com/Creality3DPrinting/CR-10/blob/master/Firmware/Marlin/Configuration.h

Official Site:
https://www.creality3donline.com/creality-5pcs-100k-ohm-ntc-3950fb-thermistors-with-cable-temperature-sensor-reprap-extruder-for-creality-3d-printer-mend-hot-bed_p0035.html

[image]

[image]

Deneteus, 2019-08-24:
Thanks Boelle. :)

boelle, 2019-08-24:
no problemo :-)

thisiskeithb, 2019-08-24:
See PR https://github.com/MarlinFirmware/Marlin/pull/15040.

boelle, 2019-08-24:
so we can "swing the axe" on this one when the PR gets merged :-D

thisiskeithb, 2019-08-24:
> so we can "swing the axe" on this one when the PR gets merged :-D

Yep.

These changes will be implemented in `bugfix-2.0.x` since `1.1.x` is no longer receiving support.

Deneteus, 2019-08-24:
Bugfix-2.0.x is what I was testing when I noticed that the thermistor value was off.

thisiskeithb, 2019-08-24:
> Bugfix-2.0.x is what I was testing when I noticed that the thermistor value was off.

Ahh. You linked `1.1.x`, so I was just making sure 😄

sl1pkn07, 2019-08-24:
then if affect the creality printers, the ender series is affected?

sl1pkn07, 2019-08-24:
ender series not, but CR-10_5S and CR-10mini yes:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/CR-10_5S/Configuration.h#L414

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/Creality/CR-10mini/Configuration.h#L423

(?)

thisiskeithb, 2019-08-24:
Looking at the Creality firmware for those machines, they should be updated as well. I'll update my PR.

closed 2019-08-25 by thinkyhead