MarlinFirmware/Marlin #15041: [BUG] bugfix.2.0.x POWER_LOSS_RECOVERY-Marlin does not react to the POWER_LOSS_PIN signal

ShaggyDog18, 2019-08-24:
### Description
Marlin bugfix-2.0.x, released on 21-AUG, does not react to the POWER_LOSS_PIN signal change.

### Steps to Reproduce
Power sensing circuit provides following signal levels at the POWER_LOSS_PIN:
- Normal operations - LOW
- Power outage - HIGH

Settings in config_adv.h :
#define POWER_LOSS_RECOVERY
#define POWER_LOSS_PIN        P1_26  // YMAX pin at SKR v1.3 board
#define POWER_LOSS_STATE    LOW

**Expected behavior:** 
Marlin should identify power loss once a power sensing pin goes from LOW to HIGH

**Actual behavior:** 
No reaction to a power sensing pin switching from LOW to HIGH

**Additional Information**
if I define POWER_LOSS_STATE as HIGH,  then printer identified power loss immediately after it starts printing ans halt at once.

vivian-ng, 2019-08-24:
> Settings in config_adv.h :
#define POWER_LOSS_RECOVERY
#define POWER_LOSS_PIN P1_26 // YMAX pin at SKR v1.3 board
#define POWER_LOSS_STATE LOW

> Expected behavior:
Marlin should identify power loss once a power sensing pin goes from LOW to HIGH

> Actual behavior:
No reaction to a power sensing pin switching from LOW to HIGH

Based on the above settings in Configuration_adv.h, the expected behavior is "Marlin should identify power loss once a power sensing pin goes from HIGH to LOW".

But it can't explain this:
> Additional Information
if I define POWER_LOSS_STATE as HIGH, then printer identifies "power loss" immediately after it starts printing and halt at once.

So some troubleshooting to help us narrow down the likely cause:
1. Test current logic.
Maintain Configuration_adv.h with:
#define POWER_LOSS_RECOVERY
#define POWER_LOSS_PIN P1_26 // YMAX pin at SKR v1.3 board
#define POWER_LOSS_STATE LOW
Build and flash firmware to board. Leave YMAX unconnected. See if the firmware boots as normal.
Then, connect YMAX to GND. See if firmware triggers a POWER_LOSS event.

2. Test logic based on power sensing circuit.
Change Configuration_adv.h with:
#define POWER_LOSS_RECOVERY
#define POWER_LOSS_PIN P1_26 // YMAX pin at SKR v1.3 board
#define POWER_LOSS_STATE HIGH
Connect YMAX to GND. Build and flash firmware. See if firmware boots as normal.
Disconnect YMAX from GND, this should drive YMAX high. See if POWER_LOSS event is triggered.

Try out these two tests and share the results so that we can check whether the firmware's logic is working as expected.

ShaggyDog18, 2019-08-24:
checked out both case. Both works. So, Marlin's logic is fine.
Checking up my Power Sense module.
Thank you!

vivian-ng, 2019-08-24:
@ShaggyDog18 Great to hear that Marlin's logic is fine.
Please close this issue once you are certain it is not an issue with Marlin. Thank you!

ShaggyDog18, 2019-08-24:
OK, root cause - bad contact of the signal wire in the connector:-(   Resolved! 
The second scenario works now.
Thank you for your support.

closed 2019-08-24 by ShaggyDog18