MarlinFirmware/Marlin #15044: I2C Controlled Additional Axis Movement Controller

jgoo9410, 2019-08-24:
### Description

I have a printer that uses 4 lead screws to move a very large bed. The lead screws are to be used to level the bed mechanically rather than through the firmware based Z levelling system. By adding a couple of I2C commands in the manual levelling routine in "marlin.cpp", I have managed to get it to level correctly. The obvious issue with this system is that it will have to handle the general movement of the Z axis during a print. I have tried modifying marlin so that when a Z move is to happen, it will use the I2C connection to send the Z target to the Z controller along with the federate(to keep the systems in sync). Currently I have added some I2C commands in the "void gcode_get_destination()" function but the timing doesn't seem to be correct.

### Steps to Reproduce

The issue is: After a homing cycle, when G1 Z100 is send immediately followed by G1 Z0, you would expect the axis to move from 0 to 100 and then back to 0. My system is 'dumb' and will immediately move towards the Z value provided as soon as it is received over I2C. This was an intentional design consideration so that the Z controller can't ever become vastly out of sync. I would like to know where I should call the function that will send the Z value out in marlin, so that it is in time with marlin moving the Z axis. i.e. it doesn't send the I2C command to go to Z0 until it has reached Z100.

### Disclaimer

I am aware this isn't technically a feature request but it does pertain to a feature in marlin that doesn't currently exist.

Grogyan, 2019-08-25:
Can you please provide more details on this configuration?

Because it sounds like you are running I2C to each of the 4 Z axis motors.

jgoo9410, 2019-08-25:
Marlin running on one arduino, connected via I2C to another arduino that controls the 4 motors. The second arduino controlling the 4 motors reacts immediately to coordinates received from the arduino running marlin. I’d like the data to be sent out only when marlin wants to make a move, not just as it’s received on the serial bus.

Grogyan, 2019-08-25:
This not optimal.
Have you looked into the BTT SKR PRO?

Generally you want to keep all that IO limited to just on one board, as I2C is quite susceptible to EMI

jgoo9410, 2019-08-26:
Unfortunately a larger board isn’t really an option. I currently have 2X motors, 2Y motors, 4E motors and 4Z motors. I have designed my own PCB hosting 8 tmc2130’s as I would like the feedback they offer. I have built the board and it works but there aren’t enough pins left to control 4 more motors.

DerAndere1, 2019-09-24:
Do the 4 additional motors have to move independent of each other? If they always have to do the same movement, why not wire them in series or in parallel to one stepper driver and beef up the stepper driver (e.g. Pololu High-Power Stepper Motor Driver 36v4), if required?

jgoo9410, 2019-09-25:
They have to move independently of each other. I used some code written by thinkyhead which worked well. The code sends a character when the printer wants to home. It then waits till there is a response from the z axis controller, indicating the homing procedure has completed. After that, using port manipulation (for speed), the z controller will read the step pulse output from marlin and mimic it to the 4 step pulses on the z controller.