MarlinFirmware/Marlin #15047: [BUG] increasing X,Y jerk past 7 will cause Z to stop moving during print

miso87, 2019-08-24:
When jerk settings for X and Y are set to value bigger than 7 either in FW or by "eg. M201 X8 Y8" the Z motor will not move and remain dead silent during a print, but Z coordinate increase with the print and X and Y move as they should. Homing Z or moving it manually through any client when not printing works fine with XY jerk values much higher than 7, the problem is during print only.
 
Another example:
If print is started with X and Y jerk lower than 7, the Z is moving and print is going well. When entering M201 X and Y bigger than 7 at any time during the print, the Z motor will immediately stop advancing, but Z coordinate does advance. XY are moving as they should with new jerk settings applied. The Z motor is not even buzzing but dead silent, the motor temperature drops as sign the Z driver is not receiving step commands anymore from the board. Trinamic debug M122 shows the Z motor/driver is standstill (stst) as well.

* Marlin 2.0 latest bugfix 
* Cartesian setup with dual Y axis
* SKR 1.3 board + 5x TMC2208 in UART mode
* Octoprint, Pronterface
* Zaxis driven by 5mm pitch ballscrew + belt reduction
----
  
* DEFAULT_AXIS_STEPS_PER_UNIT   { 160, 160, 640, 816 } 
* DEFAULT_MAX_FEEDRATE          { 200, 200, 35, 25 }
* DEFAULT_MAX_ACCELERATION      { 5000, 5000, 500, 5000 }
* microstepping 32, 32, 16, 32  X, Y, Z, E0
----
* Matching  Z acceleration to same value as XY did not help.
* Matching Z jerk to same value as XY did not help
* Lowering Z steps per unit from 1280 to 640 and Z microstepping from 32 to 16 did not help.

**Any ideas what could be the reason of this behavior or any tips what to try?**

Actual config
[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/3537544/Marlin_config.zip)

boelle, 2019-09-24:
@miso87 is the issue still there?

miso87, 2019-09-25:
> 
> 
> @miso87 is the issue still there?

Yes the issue is still there, but if needed you can mute or close the tread.

boelle, 2019-10-21:
i will test this one later today, i need to test https://github.com/MarlinFirmware/Marlin/issues/13403 and https://github.com/MarlinFirmware/Marlin/issues/14820 anyway

boelle, 2019-10-21:
i looked at my config and its set to 10, so i cant confirm this one

/**
 * Default Jerk limits (mm/s)
 * Override with M205 X Y Z E
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#define CLASSIC_JERK
#if ENABLED(CLASSIC_JERK)
  #define DEFAULT_XJERK 10.0
  #define DEFAULT_YJERK 10.0
  #define DEFAULT_ZJERK  0.2

  //#define LIMITED_JERK_EDITING        // Limit edit via M205 or LCD to DEFAULT_aJERK * 2
  #if ENABLED(LIMITED_JERK_EDITING)
    #define MAX_JERK_EDIT_VALUES { 20, 20, 0.6, 10 } // ...or, set your own edit limits
  #endif
#endif

#define DEFAULT_EJERK    2.5  // May be used by Linear Advance

boelle, 2019-10-27:
since there is no activity and i cant reproduce i will close this one

we can always reopen if others can confirm there is a issue

closed 2019-10-27 by boelle