MarlinFirmware/Marlin #15048: [BUG] FAST_PWM_FAN causes BL Touch clone failure on BTT SKR v1.3

pyunramura, 2019-08-24:
### Description
When '#define FAST_PWM_FAN' is activated in the most recent Marlin bugfix-2.0.x as of 24.07.19, doing so causes unexpected behavior with BL Touch clone (3D Touch) on BigTreeTech SKR V1.3 board.

Prior to enabling the feature, the probe responds to deploy/stow/self-test/etc. gcode commands as normal and the blue LED in the probe lights only when issued a command. With the feature enabled the fans do not exhibit PWM noise as expected, but as a result the probe no longer responds to gcode commands and the blue LED in the probe remains lit constantly. When the feature is disabled again, the probe reverts to the original working behavior. 

I am uncertain if this is a board specific issue at the moment as I only have a SKR mini-E3 DIP to test it against; which does not support FAST_PWM_FAN currently. 

I'd be happy to provide additional documentation if requested.

### Steps to Reproduce

1. Activate "#define FAST_PWM_FAN" in configuration.h using aforementioned hardware.
2. Issue deploy/stow command to probe.
3. Note if LED in probe remains lit constantly.

**Expected behavior:**

Fan should be adjustable without PWM noise and z probe should respond to gcode commands.

**Actual behavior:**

Fans are now adjustable without PWM noise, but z probe does not respond to gcode commands. Blue LED in probe remains lit continually.

#### Additional Information

* Configuration files are attached.
[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3537551/config_files.zip)

thisiskeithb, 2019-08-24:
It's a known issue with the SKR 1.3 since servos and fans share the same timer (there's another issue somewhere in the queue with more details/discussion).

You'll have to disable `FAST_PWM_FAN` if using a BLTouch.

pyunramura, 2019-08-24:
> It's a known issue with the SKR 1.3 since servos and fans share the same timer ~(there's another issue somewhere in the queue with more details/discussion)~ Edit: See #13861.
> 
> TL;DR: You'll have to disable `FAST_PWM_FAN` if using a BLTouch.

Thanks for the info, I completely missed that. I'll go ahead and close this.

closed 2019-08-24 by pyunramura

tpruvot, 2019-08-24:
on our board its the X driver, go figure why :p ... only on the antlabs, which seems very sensible to the interferences...